kiddin9/Kwrt #272: x86版本 运行系统日志报错

ideatechfr, 2021-01-23:
装在群晖VMM虚拟机，运行后在系统日志里总有下面的错误提示，请问大佬如何处理？
`Sat Jan 23 22:50:26 2021 daemon.err uhttpd[3229]: sh: smartctl: not found
Sat Jan 23 22:50:26 2021 daemon.err uhttpd[3229]: sh: blkid: not found
Sat Jan 23 22:50:26 2021 daemon.err uhttpd[3229]: sh: blkid: not found`

kunish, 2021-03-12:
Same here from a fresh installation.
Wondering if it is a issue with vmware esxi or openwrt it self.
Any idea? It's pretty annoying to me.

kunish, 2021-03-12:
Alright guys. After a little bit of digging into this issue, I think i finally figured it out. Turns out uhttpd is requiring some dependencies(smartctl and blkid) to show some disk health and identity information.

To fix this, you just simply go into system/software, install the missing packages
- smartmontools
- blkid

After logout and login again you won't see the error messages again now on. That should do the trick.

[image]
[image]

Hope it helps.

By the way, to the owner of this project, I think it is a amazing piece of work. I really enjoy using it. Keep it up!

closed 2021-12-29 by kiddin9