kiddin9/Kwrt #273: CPU占用奇高，发现都是KILLALL进程给耗完了的

ethan-www, 2021-01-25:
#
[image]
`32076  6050 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 2719 11477 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 3640 13932 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 1225  9140 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
  916  3006 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 2949 12592 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 7848  5022 root     R     1492   0%   1% [ps]
 8041 28472 root     R     1492   0%   1% [ps]
 7862  7817 root     R     1492   0%   1% [ps]
 8180 18923 root     R     1492   0%   1% [ps]
 8313 24657 root     R     1492   0%   1% [ps]
 8386 17644 root     R     1492   0%   1% [ps]
 8522 25915 root     R     1492   0%   1% [ps]
 8584 17095 root     R     1492   0%   1% [ps]
 8495 27000 root     R     1492   0%   1% [ps]
 8616   723 root     R     1492   0%   1% [ps]
 8726 24451 root     R     1492   0%   1% [ps]
 8909 19094 root     R     1492   0%   1% [ps]
 3569 12660 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
11329 18902 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 1402  9916 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 5191 13520 root     R     1172   0%   1% killall -q -9 ss-redir ssr-redir v2ray xray trojan-plus trojan-go ipt2socks naive redsocks2 kcptun-client obfs-loca
 7793 16919 root     R     1492   0%   0% [ps]
 7705 30974 root     R     1492   0%   0% [ps]
 7890  9629 root     R     1492   0%   0% [ps]
 8646 27090 root     R     1492   0%   0% [ps]
 8876 15982 root     R     1492   0%   0% [ps]
 8434 23730 root     R     1492   0%   0% [ps]
 8431 30552 root     R     1492   0%   0% [ps]
 8722 20936 root     R     1492   0%   0% [ps]
 8821 15721 root     R     1492   0%   0% [ps]
32119  8899 root     R     1552   0%   0% top
    9     2 root     SW       0   0%   0% [ksoftirqd/0]
   15     2 root     SW       0   0%   0% [ksoftirqd/1]
  707 11854 root     S     1620   0%   0% [sh]
   10     2 root     IW       0   0%   0% [rcu_sched]
 1151  9041 root     S     1612   0%   0% [sh]
 3962 17031 root     S     1588   0%   0% /bin/sh /etc/rc.common /etc/init.d/dnsmasq restart
 1550 11820 root     S     1612   0%   0% [sh]
 1016 12733 root     S     1612   0%   0% [sh]
 2443 13104 root     S     1588   0%   0% /bin/sh /etc/rc.common /etc/init.d/dnsmasq restart
`

garypang13, 2021-01-25:
奇怪, killall 怎么会没执行成功常驻后台,用的是什么设备?

ethan-www, 2021-01-25:
ikuai 内置的X86虚拟机

ethan-www, 2021-01-25:
用的这个固件
01.25-openwrt-x86-64-generic-squashfs-combined.img.gz

garypang13, 2021-01-25:
service bypass restart 重启几次试试, 看是不是特例,

ethan-www, 2021-01-25:
重启路由系统后好了，也能正常出墙，放了一下午没管，刚去看卡得SSH都打不开了
勉强转了很久后在LUCI的终端里打开TOP命令，发现还是这个问题
[image]

ethan-www, 2021-01-25:
新下载的镜像，bypass部分就执行了下面的配置命令，其他都没动
`uci set bypass.@global[0].udp_relay_server='same'
uci set bypass.@global[0].dns_mode='0'
uci set bypass.@global[0].dns_mode_o='doh'
uci set bypass.@global[0].doh_dns_o='quad9'
uci set bypass.@global[0].dns_mode_l='1'
uci set bypass.@global[0].dns_mode_d='udp'
uci set bypass.@global[0].udp_dns_d='10.1.1.1'
for i in $(seq 20);do uci -q delete bypass.@servers[];done
uci -q delete bypass.@server_subscribe[0].subscribe_url
uci add_list bypass.@server_subscribe[0].subscribe_url='xxxxxxxxxxxxxxxxxxxxxxxxxx'
uci set bypass.@server_subscribe[0].filter_words='产品名称/过期时间/到期时间/剩余流量'
uci commit bypass
/usr/share/bypass/subscribe
/etc/init.d/bypass restart
`

jerry945, 2021-02-03:
我出现类似的问题，j1900设备，非虚拟机。
OpenWrt SNAPSHOT 01.25.2021 by GaryPang / LuCI Master git-fb6e62e
但是我没看进程，我是因为没网络，后台打开非常慢，看到cpu占用100%，直接重启了，就好了。。继续观察

garypang13, 2021-02-05:
更新了bypass ,去掉了killall, 应该不会再出现这个问题了.

closed 2021-12-29 by kiddin9