kiddin9/Kwrt #274: R2S怎么开启无线功能？

Hitman47IOI, 2021-01-27:
试了很多个插件，终于找到最喜欢的了，但是唯独网络中没有无线选项，求问应该怎么操作？谢谢！

garypang13, 2021-01-27:
用的是什么无线网卡?

Hitman47IOI, 2021-01-27:
> 用的是什么无线网卡?

磊科的NW392，官方原生支持的好像

Hitman47IOI, 2021-01-27:
是要安装以下插件吗？还是怎么操作？真的太喜欢你这个固件了，精简美观

luci-app-dcwapd | git-20.108.38431-8f34e10 | 3.0 KB | Dual Channel Wi-Fi AP Daemon configuration module
-- | -- | -- | --

Hitman47IOI, 2021-01-27:
RTL8812AU
详见https://wiki.friendlyarm.com/wiki/index.php/NanoPi_R2S/zh#.E5.A6.82.E4.BD.95.E5.9C.A8.E7.BB.88.E7.AB.AF.E4.BD.BF.E7.94.A8.E5.91.BD.E4.BB.A4.E6.9F.A5.E8.AF.A2USB.E6.97.A0.E7.BA.BF.E7.BD.91.E5.8D.A1.E5.9E.8B.E5.8F.B7

Hitman47IOI, 2021-01-27:
感谢大神鼎力相助，如果能解决就太好啦，静候佳音~

garypang13, 2021-01-27:
刷别人的固件能开启这个无线吗,

Hitman47IOI, 2021-01-27:
> 刷别人的固件能开启这个无线吗,

可以的，官方的，soffchen，songchenwen以及klever1988的固件均可完美识别和运行

garypang13, 2021-01-27:
无线网卡型号太多了,我也不知道NW392对应的的是哪个驱动.

Hitman47IOI, 2021-01-27:
> 无线网卡型号太多了,我也不知道NW392对应的的是哪个驱动.

不知道是不是这个
http://www.netcoretec.com/portal/search/service_search.html

garypang13, 2021-01-27:
应该是 rtl8812au 这个,等下一版固件更新加上.

Hitman47IOI, 2021-01-27:
> 应该是 rtl8812au 这个,等下一版固件更新加上.

太感谢了，还想再问一下，更新完之后是否直接会在网络菜单下直接显示无线的选项呢？以及下一版固件是指下一次更新还是大版本更新？
感激不尽！

PassXml, 2021-01-27:
- - 我用8811cu的网卡..好像也有点问题....能开wifi..但是开加密的情况..任何设备都过不了验证...

楼主注意下

Hitman47IOI, 2021-01-27:
> * * 我用8811cu的网卡..好像也有点问题....能开wifi..但是开加密的情况..任何设备都过不了验证...
> 
> 楼主注意下

你刷写完固件，网络那里就会自动显示“无线”这个选项吗？还是说要经过一定操作才会出现？

PassXml, 2021-01-28:
> > * * 我用8811cu的网卡..好像也有点问题....能开wifi..但是开加密的情况..任何设备都过不了验证...
> > 
> > 楼主注意下
> 
> 你刷写完固件，网络那里就会自动显示“无线”这个选项吗？还是说要经过一定操作才会出现？

直接出现...

Hitman47IOI, 2021-01-28:
> > > * * 我用8811cu的网卡..好像也有点问题....能开wifi..但是开加密的情况..任何设备都过不了验证...
> > > 
> > > 楼主注意下
> > 
> > 
> > 你刷写完固件，网络那里就会自动显示“无线”这个选项吗？还是说要经过一定操作才会出现？
> 
> 直接出现...

对呀，我就觉得奇怪了，按道理即使没识别成功也应该有显示这个选项的呀，为什么我的木有，是只刷nanopi-r2s-squashfs-sysupgrade.img.gz就行了嘛？还是这个“nanopi-r2s_sha256sums ”也要刷？

PassXml, 2021-01-28:
> nanopi-r2s_sha256sums

nanopi-r2s_sha256sums作为校验用的...你lsusb 看下识别usb万卡了没有

Hitman47IOI, 2021-01-28:
> > nanopi-r2s_sha256sums
> 
> nanopi-r2s_sha256sums作为校验用的...你lsusb 看下识别usb万卡了没有

没有，其他固件就可以

Hitman47IOI, 2021-02-01:
> 应该是 rtl8812au 这个,等下一版固件更新加上.

求教大神，本次更新中，“网络”菜单下依旧没有“无线”选项，请问是本人操作的问题吗？

epenli, 2021-02-16:
rtl8812au  还是没有加上 库里没有

closed 2021-12-29 by kiddin9