kiddin9/Kwrt #281: 无法安装 kmod-fs-ntfs

xbingc, 2021-01-31:
Intel(R) Celeron(R) CPU 3965U @ 2.20GHz
固件版本 | OpenWrt SNAPSHOT 11.29.2020 by GaryPang / LuCI Master git-b3bbec9
内核版本 | 5.4.80

手贱卸载了 kmod-fs-ntfs，发现无论如何都装不回去了
使用 opkg install 提示：

Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Unknown package 'kmod-fs-ntfs'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-ntfs found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-fs-ntfs.

从很多地方下载过多个版本的IPK，想本地安装也失败了，使用 --force-overwrite无效
测试的版本包括 kernel 一致， Architecture 为 x86_64的版本

想请教大佬，除了升级固件外，还有什么办法把它装回去吗？我挂载的移动硬盘需要用
上次升级固件一些插件配置丢了，所以这次想能不升级就先不升级了

garypang13, 2021-01-31:
用 --force-depends  而不是 -force-overwrite

xbingc, 2021-01-31:
> 用 --force-depends 而不是 -force-overwrite

这么快得到回复~太好了
我试了一下，不知道是不是语法不对，还是没有成功

/tmp# opkg install /tmp/kmod-fs-ntfs_5.4.80-1_x86_64.ipk --force-depends
/tmp# opkg install --force-depends /tmp/kmod-fs-ntfs_5.4.80-1_x86_64.ipk
/tmp# opkg --force-depends install /tmp/kmod-fs-ntfs_5.4.80-1_x86_64.ipk

用了三种格式，返回的是相同的错误 （主楼的信息）

garypang13, 2021-01-31:
[image]

这样就代表成功了,

xbingc, 2021-01-31:
[image]
试了一下，执行结果如图……有其它地方不对吗

garypang13, 2021-01-31:
安装前先 opkg update 更新下插件列表
最好是更新下固件,升级固件前你可以先备份下配置, 一般是不会丢失配置的,如果有哪个插件配置丢失可以和我反馈.

xbingc, 2021-01-31:
> 安装前先 opkg update 更新下插件列表
> 最好是更新下固件,升级固件前你可以先备份下配置, 一般是不会丢失配置的,如果有哪个插件配置丢失可以和我反馈.

update已经运行过了，看来只能升级固件了
上一次升级后是OpenClash的配置丢失了。

xbingc, 2021-02-20:
> 安装前先 opkg update 更新下插件列表
> 最好是更新下固件,升级固件前你可以先备份下配置, 一般是不会丢失配置的,如果有哪个插件配置丢失可以和我反馈.

过完节来继续折腾了，继续请教
刚才用固件里的在线升级，从5.4.80 到 5.4.99
升级完毕后发现OPENCLASH不见了，这个和上一次使用在线升级的情况是一样的
想问下这样还能通过升级前备份的设置恢复吗？还是必须重新安装一次OPENCLASH了？

closed 2021-12-29 by kiddin9