kiddin9/Kwrt #284: R2S使用可道云无法启用nginx和php

HalcyonAzure, 2021-02-01:
操作步骤:
1.将最新版(2月1日)固件刷入SD卡,并且重置删除之前所有操作
2.在软件包安装nginx和php7后，安装kodexplorer英文原版
3.在NAS->Kodexplorer中设置端口为8081,并且修改好文件路径和项目路径，保存并应用
4.点击检查更新以后得到更新1.15的提示，再次点击更新并且下载后勾选启用，再次保存并应用

出现的问题:
nginx和php的状态依旧是未运行，无法使用8081端口进行登入。
通过ssh登入以后输入“nginx”有反应，同时通过SCP查看设定的项目下载目录也确实下载了可道云的文件

正常情况:
全部设置完毕，勾选启用以后，点击保存并应用，nginx和php会自动启用，并且能够访问可道云网站

日志：
在点击“检查更新“并”下载“完毕更新以后，到勾选启动并且点击保存并应用的过程中，通过ssh的logread -f
读取到的日志:https://pastebin.ubuntu.com/p/vSvZ46N9JN/
其中10.0.0.205为SCP登入的log

备注:
1.安装可道云的时候R2S提示依赖库有一定问题，出现以下提示
        已安装的软件包 nginx-ssl-util 版本不兼容，要求 > = 1.5-1 而 1.5-1 已安装。
        已安装的软件包 nginx-util 版本不兼容，要求 > = 1.4-2 而 1.5-1 已安装。
        依赖的软件包 nginx-ssl-util (>=1.5-1) 在所有仓库都未提供。
2.在其他固件上用大致相同的流程在R2S上直接安装可道云的ipk可以正常进入并运行

aenron, 2021-02-02:
r4s也有这个问题

garypang13, 2021-02-02:
已修复,卸载luci-app-kodexplorer后重装

HalcyonAzure, 2021-02-02:
刚刚重置之后再次重复上述反馈的步骤进行安装，问题依旧存在，使用 logread -f 得到的内容和昨天大致没有区别。想问一下是不是需要自己提前手动安装一些依赖库？还是说理论上来说直接安装luci-app-kodexplorer后会在安装过程中自动安装好对应依赖库需要的文件？

Ps：之前偶然用另外下载的其他地方的kodexplorer在路由器上成功运行过一次，但那次运行的时候可能自己有杂七杂八装过什么依赖库，并且第二天早上起来的时候发现已经自动关闭了，之后无论如何想要复现那次运行都没有成功过，包括排查是否是语言问题，是否是其他插件导致的端口冲突，是否是可道云本身没有下载等

目前自动更新得到的可道云版本是1.15，通过查阅别人的一些帖子发现有的是4.40？这两个版本号不同会有影响吗？

另外今天早上看见系统有2月2号的更新，就在更新之前和更新之后均尝试过恢复出厂设置并安装，均失败。
是否有其他需要的log可以查阅并上传？

感谢作者回复。

garypang13, 2021-02-02:
后台 启动项 中 检查下kodexplorer如果是禁用,改为启用后重启下路由

HalcyonAzure, 2021-02-02:
成功了！十分感谢！

closed 2021-02-02 by HalcyonAzure