kiddin9/Kwrt #293: R2S 版本 可否完善一下 lxc 的编译底层配置 现在安装是没办法工作的 多谢

epenli, 2021-02-05:
R2S 版本 可否完善一下 lxc 的编译底层配置 现在安装是没办法工作的 多谢

garypang13, 2021-02-08:
luci-app-lxc在线安装后没办法工作吗,我没用过lxc, lxc的编译底层配置没挺明白具体指的是什么.

epenli, 2021-02-16:
用这个包跑一下测试 lxc-checkconfig

Cgroup v2 mount points: 
/sys/fs/cgroup

Cgroup v1 systemd controller: missing
Cgroup v1 freezer controller: missing
Cgroup namespace: required
Cgroup device: missing
Cgroup sched: enabled
Cgroup cpu account: enabled
Cgroup memory controller: enabled
Cgroup cpuset: enabled

--- Misc ---
Veth pair device: enabled, loaded
Macvlan: enabled, loaded
Vlan: enabled, not loaded
Bridges: enabled, not loaded
Advanced netfilter: enabled, not loaded
CONFIG_NF_NAT_IPV4: missing
CONFIG_NF_NAT_IPV6: missing
CONFIG_IP_NF_TARGET_MASQUERADE: missing
CONFIG_IP6_NF_TARGET_MASQUERADE: missing
CONFIG_NETFILTER_XT_TARGET_CHECKSUM: missing
CONFIG_NETFILTER_XT_MATCH_COMMENT: enabled, loaded
FUSE (for use with lxcfs): enabled, loaded

--- Checkpoint/Restore ---
checkpoint restore: missing
CONFIG_FHANDLE: enabled
CONFIG_EVENTFD: enabled
CONFIG_EPOLL: enabled
CONFIG_UNIX_DIAG: missing
CONFIG_INET_DIAG: missing
CONFIG_PACKET_DIAG: missing
CONFIG_NETLINK_DIAG: missing
File capabilities: 

Note : Before booting a new kernel, you can check its configuration
usage : CONFIG=/path/to/config /usr/bin/lxc-checkconfig

epenli, 2021-02-16:
> luci-app-lxc在线安装后没办法工作吗,我没用过lxc, lxc的编译底层配置没挺明白具体指的是什么.

Cgroup v1 systemd controller: missing
Cgroup v1 freezer controller: missing
Cgroup namespace: required

这三个都是红色的 必须内核配置 大神看什么时候有空捣鼓一下？

closed 2021-12-29 by kiddin9