kiddin9/Kwrt #295: 这个samba3.X的版本没有Ui吗？

lzqd, 2021-02-06:


garypang13, 2021-02-08:
samba3.X请用ksmbd替代,

lzqd, 2021-02-08:
> samba3.X请用ksmbd替代，

这个ui使用的版本和中文包的版本不一致是什么情况### 
[image]
[image]

lzqd, 2021-02-08:
> samba3.X请用ksmbd替代,

samba4也存在ui和语言包版本不匹配，然后语言包问题导致汉化无法使用

garypang13, 2021-02-08:
1. 确实是否有安装对应的语言包
2. ctrl+F5 刷新浏览器缓存

lzqd, 2021-03-18:
[image]
[image]
UI和翻译版本还是不一致，为什么会这样？

closed 2021-12-29 by kiddin9