kiddin9/Kwrt #328: luci-app-openvpn-server无法生成证书

IveanEx, 2021-03-20:
我的路由器是红米AC2100，大概从一个月前开始luci-app-openvpn-server就无法生成证书了，不管直接点Web页面的生成证书还是执行脚本好像都没效果……

closed 2021-12-29 by kiddin9

huanglin199009, 2022-06-30:
我的也是这样，求解决方法

huanglin199009, 2022-06-30:
openvpn: OpenVPN 2.5.5 mipsel-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]

huanglin199009, 2022-06-30:
NOTE: the current --script-security setting may allow this configuration to call user-defined scripts

huanglin199009, 2022-06-30:
OpenSSL: error:02001002:system library:fopen:No such file or directory

huanglin199009, 2022-06-30:
OpenSSL: error:2006D080:BIO routines:BIO_new_file:no such file

huanglin199009, 2022-06-30:
NOTE: your local LAN uses the extremely common subnet address 192.168.0.x or 192.168.1.x.  Be aware that this might create routing conflicts if you connect to the VPN server from public locations such as internet cafes that use the same subnet.