kiddin9/Kwrt #342: PVE虚拟机X86版cpu占用奇高

coLATin, 2021-04-01:
[image]

这可是T9900的U啊，首先显示不正确，另外cpu占用奇高，之前貌似5%的时候都不多，现在始终在30-40%。。。。

garypang13, 2021-04-01:
命令行 htop 查看时哪些进程占用的CPU

CPU显示不正确时因为用了PVE,显示的就是虚拟出来的CPU信息.

coLATin, 2021-04-01:
[image]

coLATin, 2021-04-01:
用工具按照cpu从高到底捋，也并没有过分的占用....奇怪了

coLATin, 2021-04-01:
[image]
并且是持续的，内存也吃了2g多.....

coLATin, 2021-04-01:
[image]
bypass的插件是主要原因，杀进程后25%（也是高），启用bypass进程后，就一直在40%多了

283915025, 2021-04-09:
虚拟机CPU选host，不然效率太低

closed 2021-12-29 by kiddin9