kiddin9/Kwrt #350: Qos over Nftables 功能没有生效

ideatechfr, 2021-04-08:
x86 系统，版本升级到OpenWrt 21.02-SNAPSHOT 04.03.2021 ，因为用户比较多，所以配置了Qos over Nftables功能
[image]
应用后，实测客户端的速度没有被限制。

另外，在状态里实时信息，里面有个实时速率显示，我链接了3个终端进行网速测试，但这里没有显示，能不能帮忙看一下什么问题
[image]

garypang13, 2021-04-08:
关闭 turbo acc网络加速中的 Shortcut-FE 流量分载 后再试试.

ideatechfr, 2021-04-08:
已经关了，限速还是没有生效

[image]

garypang13, 2021-04-08:
[image]

ideatechfr, 2021-04-08:
设置为静态地址过滤
[image]
实测网速
[image]

garypang13, 2021-04-08:
确实有问题,等我后续再做测试和修复.

ideatechfr, 2021-04-09:
好的，谢谢！

另外还发现Rate实时速率显示功能，里面的IP不是唯一，有很多重复IP显示在列表里。因为现在测试是只有10多个IP链接，但列表已经拉的很长。 麻烦帮忙看看是否可以解决
[image]

iwrt, 2021-12-16:
nft确实没有用

closed 2021-12-29 by kiddin9