kiddin9/Kwrt #372: 关于软件包安装失败疑惑

0x2CA, 2021-04-22:
我想使用UnblockNeteaseMusic、luci-app-homebridge之类的，状态提示未运行，有安装按钮，我点击打印如下
````
Begin Checking Environment--------------
OPKG update -------------------------
Downloading https://op.supes.top/snapshots/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://op.supes.top/packages/mipsel_24kc//Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_custom
OPKG update success
OPKG update finished ----------------
Begin Checking Node environment-----------------------
node already installed
try install node-npm
Unknown package 'node-npm'.
Collected errors:
 * opkg_install_cmd: Cannot install package node-npm.
node-npm install failed
```
提示安装`node-npm`失败，好像是安装npm，我发现固件是有node的，但是没有npm

Magic-Xin, 2021-04-23:
node-npm 目前只支持到 OpenWrt-19.07.0，等上游更新吧

0x2CA, 2021-04-24:
额那其实要很久的把

closed 2021-12-29 by kiddin9