kiddin9/Kwrt #382: 最新版编译失败,URL失效？

RoyLi-Pvt, 2021-05-16:
日志如下:
make[3]: Entering directory '/home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/feeds/custom/autocore'
mkdir -p /home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/dl
SHELL= flock /home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/tmp/.cpuinfo.flock -c '  	/home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/scripts/download.pl "/home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/dl" "cpuinfo" "skip" "cpuinfo" "https://raw.githubusercontent.com/immortalwrt/immortalwrt/master/package/lean/autocore/files/arm"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/immortalwrt/immortalwrt/master/package/lean/autocore/files/arm/cpuinfo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/cpuinfo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/cpuinfo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/cpuinfo
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:75: /home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/dl/cpuinfo] Error 2
make[3]: Leaving directory '/home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/feeds/custom/autocore'
time: package/feeds/custom/autocore/x86/compile#0.27#0.11#1.83
    ERROR: package/feeds/custom/autocore failed to build (build variant: x86).
make[2]: *** [package/Makefile:114: package/feeds/custom/autocore/compile] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt'
make[1]: *** [package/Makefile:108: /home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt'
make: *** [/home/runner/work/OpenWrt-Snap-CI/OpenWrt-Snap-CI/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

我检查了下immortalwrt/immortalwrt/master/package/lean/autocore/files/arm 确实是不存在的
相关的文件应该在immortalwrt/immortalwrt/tree/master/package/emortal/autocore/files/arm
但是我并没有找到相关的链接引用在哪里,此外为什么我编译x86却会下载arm的文件呢？

garypang13, 2021-05-16:
已修复.

RoyLi-Pvt, 2021-05-16:
额 能说下在哪里修复的吗?🤣

RoyLi-Pvt, 2021-05-16:
额 能说一下具体是哪里引用了这个地址嘛😂我注意到实际的修复内容是提交到了garypang13/my-packages但是我依然没找到哪里引用了garypang13/my-packages，多谢~

RoyLi-Pvt, 2021-05-17:
此外 我Fork后用Action编译似乎还是会报错,不知道是不是昨天GitHubAction出问题导致的,我再试试

guanzi008, 2021-05-22:
[ 22%] Built target cmcurl
make[6]: Entering directory '/home/xuhao/openwrt/build_dir/host/cmake-3.19.1'
make[6]: Leaving directory '/home/xuhao/openwrt/build_dir/host/cmake-3.19.1'
make[6]: Entering directory '/home/xuhao/openwrt/build_dir/host/cmake-3.19.1'
[ 23%] Linking C executable curltest
/usr/bin/ld: lib/libcmcurl.a(openssl.c.o): in function `Curl_ossl_sha256sum':
openssl.c:(.text+0x61): undefined reference to `EVP_MD_CTX_create'
/usr/bin/ld: openssl.c:(.text+0x9c): undefined reference to `EVP_MD_CTX_destroy'
/usr/bin/ld: lib/libcmcurl.a(openssl.c.o): in function `Curl_ossl_md5sum':
openssl.c:(.text+0x101): undefined reference to `EVP_MD_CTX_create'
/usr/bin/ld: openssl.c:(.text+0x13c): undefined reference to `EVP_MD_CTX_destroy'
/usr/bin/ld: lib/libcmcurl.a(openssl.c.o): in function `Curl_ossl_cleanup':
openssl.c:(.text+0x919): undefined reference to `EVP_cleanup'
/usr/bin/ld: openssl.c:(.text+0x91e): undefined reference to `ENGINE_cleanup'
/usr/bin/ld: openssl.c:(.text+0x923): undefined reference to `ERR_free_strings'
/usr/bin/ld: openssl.c:(.text+0x92f): undefined reference to `CONF_modules_free'
/usr/bin/ld: lib/libcmcurl.a(openssl.c.o): in function `ossl_new_session_cb':
openssl.c:(.text+0x11dc): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: openssl.c:(.text+0x1204): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: lib/libcmcurl.a(openssl.c.o): in function `Curl_ossl_init':
openssl.c:(.text+0x1331): undefined reference to `SSL_load_error_strings'
/usr/bin/ld: openssl.c:(.text+0x1336): undefined reference to `SSL_library_init'
/usr/bin/ld: openssl.c:(.text+0x1349): undefined reference to `OPENSSL_add_all_algorithms_noconf'
/usr/bin/ld: openssl.c:(.text+0x137c): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: openssl.c:(.text+0x139c): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: lib/libcmcurl.a(openssl.c.o): in function `servercert':
openssl.c:(.text+0x2714): undefined reference to `sk_num'
/usr/bin/ld: openssl.c:(.text+0x2767): undefined reference to `sk_value'
/usr/bin/ld: openssl.c:(.text+0x2b68): undefined reference to `sk_num'
/usr/bin/ld: openssl.c:(.text+0x2be0): undefined reference to `sk_value'
/usr/bin/ld: openssl.c:(.text+0x2e9d): undefined reference to `sk_num'
/usr/bin/ld: openssl.c:(.text+0x2f04): undefined reference to `sk_num'
/usr/bin/ld: openssl.c:(.text+0x2f16): undefined reference to `sk_value'
/usr/bin/ld: lib/libcmcurl.a(openssl.c.o): in function `ossl_connect_step1':
openssl.c:(.text+0x45b1): undefined reference to `SSL_CTX_set_min_proto_version'
/usr/bin/ld: openssl.c:(.text+0x4607): undefined reference to `SSL_CTX_set_max_proto_version'
/usr/bin/ld: openssl.c:(.text+0x4972): undefined reference to `sk_pop_free'
/usr/bin/ld: openssl.c:(.text+0x4f16): undefined reference to `SSL_CTX_set_min_proto_version'
/usr/bin/ld: openssl.c:(.text+0x4f24): undefined reference to `SSL_CTX_set_max_proto_version'
/usr/bin/ld: openssl.c:(.text+0x5348): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: openssl.c:(.text+0x5905): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: openssl.c:(.text+0x5e69): undefined reference to `sk_pop'
/usr/bin/ld: openssl.c:(.text+0x5ea7): undefined reference to `sk_num'
/usr/bin/ld: openssl.c:(.text+0x5ed9): undefined reference to `sk_pop_free'
/usr/bin/ld: openssl.c:(.text+0x606f): undefined reference to `SSL_CTX_clear_chain_certs'
/usr/bin/ld: openssl.c:(.text+0x60a3): undefined reference to `SSL_CTX_add0_chain_cert'
collect2: error: ld returned 1 exit status
make[6]: *** [Utilities/cmcurl/CMakeFiles/curltest.dir/build.make:108: Utilities/cmcurl/curltest] Error 1
make[6]: Leaving directory '/home/xuhao/openwrt/build_dir/host/cmake-3.19.1'
make[5]: *** [CMakeFiles/Makefile2:785: Utilities/cmcurl/CMakeFiles/curltest.dir/all] Error 2
make[5]: Leaving directory '/home/xuhao/openwrt/build_dir/host/cmake-3.19.1'
make[4]: *** [Makefile:182: all] Error 2
make[4]: Leaving directory '/home/xuhao/openwrt/build_dir/host/cmake-3.19.1'
make[3]: *** [Makefile:39: /home/xuhao/openwrt/build_dir/host/cmake-3.19.1/.built] Error 2
make[3]: Leaving directory '/home/xuhao/openwrt/tools/cmake'
time: tools/cmake/compile#0.13#0.01#0.14
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:160: tools/cmake/compile] Error 1
make[2]: Leaving directory '/home/xuhao/openwrt'
make[1]: *** [tools/Makefile:156: /home/xuhao/openwrt/staging_dir/host/stamp/.tools_compile_yyynyynnyyynyyyyyynyynnyyyynyyyyyyyyyyyyyyyynynnyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/xuhao/openwrt'
make: *** [/home/xuhao/openwrt/include/toplevel.mk:230: world] Error 2

closed 2021-12-29 by kiddin9