kiddin9/Kwrt #423: [内网跑不满速度] 麻烦dalao看下 ikuai下测试过是满速的

BirtyAs, 2021-07-02:
网卡 I350-T4 直通 1000MBP/S /全双工 线是CAT6
op版本 OpenWrt 21.02-SNAPSHOT 06.27.2021 by GaryPang / LuCI openwrt-21.02 branch git-d67e0da

你好，dalao~
前几天发现固件跑不满速度 百度查了下似乎是固件问题 麻烦看看呗

谢谢！

下面是iperf3 的测试结果。
Accepted connection from 10.0.0.206, port 62719
[  5] local 10.0.0.1 port 5201 connected to 10.0.0.206 port 62720
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  47.7 MBytes   400 Mbits/sec    0    210 KBytes
[  5]   1.00-2.00   sec  46.3 MBytes   389 Mbits/sec    0    210 KBytes
[  5]   2.00-3.00   sec  46.3 MBytes   389 Mbits/sec    0    210 KBytes
[  5]   3.00-4.00   sec  46.7 MBytes   392 Mbits/sec    0    210 KBytes
[  5]   4.00-5.00   sec  46.3 MBytes   388 Mbits/sec    0    210 KBytes
[  5]   5.00-6.00   sec  47.0 MBytes   394 Mbits/sec    0    221 KBytes
[  5]   6.00-7.00   sec  48.3 MBytes   405 Mbits/sec    0    221 KBytes
[  5]   7.00-8.00   sec  48.7 MBytes   409 Mbits/sec    0    221 KBytes
[  5]   8.00-9.00   sec  49.9 MBytes   418 Mbits/sec    0    221 KBytes
[  5]   9.00-10.00  sec  49.7 MBytes   417 Mbits/sec    0    221 KBytes
[  5]  10.00-10.02  sec   941 KBytes   492 Mbits/sec    0    221 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.02  sec   478 MBytes   400 Mbits/sec    0             sender
-----------------------------------------------------------
Server listening on 5201 (test #2)
-----------------------------------------------------------
Accepted connection from 10.0.0.206, port 56509
[  5] local 10.0.0.1 port 5201 connected to 10.0.0.206 port 56510
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  49.8 MBytes   417 Mbits/sec    0    265 KBytes
[  5]   1.00-2.00   sec  46.3 MBytes   389 Mbits/sec    0    228 KBytes
[  5]   2.00-3.00   sec  47.2 MBytes   396 Mbits/sec    0    214 KBytes
[  5]   3.00-4.00   sec  46.3 MBytes   389 Mbits/sec    0    180 KBytes
[  5]   4.00-5.00   sec  46.3 MBytes   389 Mbits/sec    0    220 KBytes
[  5]   5.00-6.00   sec  47.4 MBytes   398 Mbits/sec    0    208 KBytes
[  5]   6.00-7.00   sec  48.7 MBytes   409 Mbits/sec    0    222 KBytes
[  5]   7.00-8.00   sec  48.7 MBytes   409 Mbits/sec    1    217 KBytes
[  5]   8.00-9.00   sec  50.5 MBytes   424 Mbits/sec    0    234 KBytes
[  5]   9.00-10.00  sec  50.2 MBytes   421 Mbits/sec    0    222 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   481 MBytes   404 Mbits/sec    1             sender

garypang13, 2021-07-02:
@BirtyAs 同样的测试方法下,有测试过其他固件么,

BirtyAs, 2021-07-02:
> @BirtyAs 同样的测试方法下,有测试过其他固件么,

Hi,
     经过测试 OpenWrt官方固件是没有问题的。
        所用固件版本：OpenWrt 21.02.0-rc3 r16172-2aba3e9784 / LuCI openwrt-21.02 branch git-21.163.64918-ba57ec5
               来源：https://downloads.openwrt.org/releases/21.02.0-rc3/targets/x86/generic/openwrt-21.02.0-rc3-x86-generic-generic-ext4-combined-efi.img.gz

以下是官方固件的iperf3测速日志:

PS D:\iperf-3.1.3-win64> .\iperf3.exe -c 192.168.2.1 -R
Connecting to host 192.168.2.1, port 5201
Reverse mode, remote host 192.168.2.1 is sending
[  4] local 192.168.2.206 port 63016 connected to 192.168.2.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   113 MBytes   949 Mbits/sec
[  4]   1.00-2.00   sec   113 MBytes   949 Mbits/sec
[  4]   2.00-3.00   sec   113 MBytes   949 Mbits/sec
[  4]   3.00-4.00   sec   113 MBytes   949 Mbits/sec
[  4]   4.00-5.00   sec   113 MBytes   949 Mbits/sec
[  4]   5.00-6.00   sec   113 MBytes   949 Mbits/sec
[  4]   6.00-7.00   sec   113 MBytes   949 Mbits/sec
[  4]   7.00-8.00   sec   113 MBytes   949 Mbits/sec
[  4]   8.00-9.00   sec   113 MBytes   949 Mbits/sec
[  4]   9.00-10.00  sec   113 MBytes   949 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  1.11 GBytes   950 Mbits/sec    0             sender
[  4]   0.00-10.00  sec  1.11 GBytes   949 Mbits/sec                  receiver

closed 2021-07-06 by BirtyAs