kiddin9/Kwrt #438: 07.26的版本在多网口情况下，br-lan桥接多个网口时系统重启后只有第一个LAN网卡会生效，重启接口后正常

wang745125665, 2021-07-26:
我设备是六个口 ETH0~6，我将ETH0设置为WLAN，ETH1~5 设置为br-lan网桥，可以正常使用，但是关机后下一次开机，只有ETH1可以启动，手动重启br-lan的接口后，其他接口又能够正常使用，这样相当于我路由器每次重启后都得进后台手动重启接口，系统为新刷，07.26的版本，重置系统还是会这样

[image]
[image]
[image]
[image]

wang745125665, 2021-07-26:
@kiddin9 用过，但是之前用的设备没有这么多网口，不确定是否会出现这个问题，这个是今天刷的新的设备，准备用做公司主路由器

kiddin9, 2021-07-26:
刷老版本试试,我是双网口,没法测试.

wang745125665, 2021-07-26:
@kiddin9 我重新刷了0627的版本，多个网口桥接的lan都能够正常启动了，可能就是后续的某个版本中出现了bug

kiddin9, 2021-07-26:
嗯嗯, 07.26有多个用户反馈有这个问题了, 07.21版没有问题.

wang745125665, 2021-07-27:
@kiddin9  请问这个需要多久能够修复呢，目前我换乘0627的版本来使用，但是安装插件的时候，很多会提示 架构不匹配
> Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-pcompress found, but incompatible with the architectures configured

wang745125665, 2021-07-27:
@kiddin9 作者你好，我看到最新版本为07.27，有解决这个问题不

kiddin9, 2021-07-27:
新版修复了,
提示 架构不匹配可以无视,你如果在后台软件包菜单中安装就不会有这些提示.

wang745125665, 2021-07-27:
@kiddin9 我是0627版本中下载比如samba4的中文语言包 提示架构不匹配，但是之前0726版本时是可以使用的，现有0627版本导致安装不了中文语言包，明天我升级最新试试

closed 2021-07-28 by wang745125665