kiddin9/Kwrt #446: 今天8月4日是不是对bypass有什么更新，导致bypass无法使用了

wang745125665, 2021-08-04:
我本身订阅链接正常，今天看到无法连上谷歌了，看了下之前的节点类型变成了v2ray，以前是获取订阅得到的是vmess，固件我是开启了软件自动更新的

learn2s, 2021-08-04:
同样问题，一直提示信息  `Main Node: Can't find V2RAY program, start failed!`

kiddin9, 2021-08-04:
是我更新过bypass的订阅文件,导致自动更新订阅的时候节点类型识别错了.

learn2s, 2021-08-04:
手动改类型也不行，请教大大如何解决

kiddin9, 2021-08-04:
手动改为vless类型也不行吗,

wang745125665, 2021-08-04:
@kiddin9 @learn2s  改类型不行，我已先换成ssr plus了，这个插件是稳定可用的

learn2s, 2021-08-04:
改类型确实不行，刚也换了 ssr plus 后可用

kiddin9, 2021-08-04:
修复了,去软件包中更新bypass下,然后点击更新所有订阅服务器节点

learn2s, 2021-08-04:
实测已正常，感谢大大

wang745125665, 2021-08-04:
@kiddin9 

我的依旧不行，在更新的时候 错误栏有  
```
Unknown command
Too many arguments
Collected errors:
```
的提示，然后虽然更新订阅后，类型正确了，但是依旧无法正常翻墙

是否是因为 bypass跟ssr plus 有很多组件相同，导致冲突了？我将bypass，ssr plus一起卸载，再重装bypass，依旧无法正常使用

我固件版本是07.27.2021，不知是否跟不是最新固件也有关系，固件内的软件包更新碰到的各种问题不少

learn2s, 2021-08-04:
我没有卸载，只是停用了 ssr plus，删除节点后重新更新节点后就可以，日志也没有报错了

wang745125665, 2021-08-04:
@learn2s 我开始的时候也是跟你一样，只停用了ssr plus，不过我在更新的时候，错误信息里面就有那个，你是不是最新07.31的固件？
我感觉这固件经常有各种错误，特别是软件包管理，不知道是不是因为不是最新，因为我固件是给很多人用，不能随意更新，更新期间路由器得断网重启，之前没关软件的自动更新，今天就出问题了

wang745125665, 2021-08-04:
安装或更新的时候都有这个错误
[image]
节点测试是通过的，但是不能访问谷歌，用ssr plus正常
[image]

learn2s, 2021-08-04:
你的 TCP 未运行就不对，看看日志？

kiddin9, 2021-08-04:
确实还有问题,已经修复了, 卸载bypass后重装,再更新订阅.

wang745125665, 2021-08-04:
@kiddin9 你已经更新了吗，我这边点了更新列表后，看到bypass的版本没有变化，然后卸载重装bypass，再去更新订阅，服务器节点都获取不到了，下面是日志

2021-08-04 17:38:47 Subscribe       : 订阅失败, 恢复服务
2021-08-04 17:38:47 Subscribe       : 更新失败，没有可用的节点信息
2021-08-04 17:38:47 Subscribe       : https://xxxx: 获取内容为空
2021-08-04 17:38:45 Subscribe       : 服务正在暂停

Jim-CN, 2021-10-09:
P.S 已重新卸载后重新安装

Jim-CN, 2021-10-09:
已安装版本比可更新版本还要高，无法升级，高版本是openwrt更新时安装上去的
[image]
[image]

closed 2021-12-29 by kiddin9