kiddin9/Kwrt #448: 新版本编译时不能用SSH自行配置固件了吗？

apc337, 2021-08-09:
新版本的/.github/workflows/repo-dispatcher.yml去掉了SSH手动配置的相关代码
  workflow_dispatch:
    inputs:
      ssh:
        description: 'SSH connection to Actions'
        required: false
        default: 'true'

现在自动跳过SSH配置了，我想手动配置一些代码，该怎么办？
谢谢啦！

kiddin9, 2021-08-09:
使用Readme中介绍的方式2油猴脚本的触发方式. 只需要在type中加上ssh 就会触发ssh
[image]

apc337, 2021-08-09:
请问这个Type在哪里啊？我没有找到。

kiddin9, 2021-08-09:
安装好油猴脚本后仓库页面就会看到这个按钮.
[image]
点击就会弹出对话框.

apc337, 2021-08-09:
请问弹出的第三个对话框Client Payload (in JSON)，这里填什么？我空着直接点了下一步，然后克隆源代码这一步就出错了

kiddin9, 2021-08-09:
如果是x86的话就填 
[image]
r2s就填 {"target": "nanopi-r2s"}

apc337, 2021-08-09:
谢谢

a717445834, 2021-09-01:
> 使用Readme中介绍的方式2油猴脚本的触发方式. 只需要在type中加上ssh 就会触发ssh
> [image]

现在只会让填token,请问要如何填写

kiddin9, 2021-09-01:
你点错按钮了, 要点 Repo Dispatch 按钮才能ssh

a717445834, 2021-09-01:
> 你点错按钮了

> 你点错按钮了

对不起,我刚刚开始用这个,不是太熟,试了几次,现在已经再编译了,谢谢大佬

closed 2021-12-29 by kiddin9