kiddin9/Kwrt #455: 友善弄出来了个R2S的换网卡版本R2C，会加入支持么

Mioscout, 2021-08-17:
如题，把R2S的RTL8211E换成了YT8521S，价格勉强接近到当时R2S第一批的价格
但是openwrt没有内置这个新网卡的驱动。等于现在就官方那个固件能用。
http://wiki.friendlyarm.com/wiki/index.php/NanoPi_R2C/zh
这个是官方的wiki

kiddin9, 2021-08-17:
@Mioscout 看情况,多几个人有需求再考虑支持.

Mioscout, 2021-08-17:
> 
> 
> @Mioscout 看情况,多几个人有需求再考虑支持.

好的，我找到了驱动，也放在这里吧
https://github.com/friendlyarm/kernel-rockchip/commit/5d6862cc5eac1679d7a4ef388f7c9bbc70e76770#diff-56a39dd90561756d31c1341807e252fa7340a1c8237ae88ef5130cfdb887cda8

Falld0wn18, 2021-10-19:
> 如题，把R2S的RTL8211E换成了YT8521S，价格便宜了六十多块钱 但是openwrt没有内置这个新网卡的驱动。等于现在就官方那个固件能用。 http://wiki.friendlyarm.com/wiki/index.php/NanoPi_R2C/zh 这个是官方的wiki

昨天看了下某宝的R2S，已经269起步了，所以买了R2C。
看了官网对比满载普遍是比R2S高了几度，请问你那边日常使用会死机吗？

closed 2021-12-29 by kiddin9