kiddin9/Kwrt #461: openvpn客户端配置(不属于固件问题)

minkaka, 2021-08-28:
我从ikuai的openVPN服务端导出的配置文件，导入到[openvpn官方客户端](https://openvpn.net/vpn-client/)后，输入用户名和密码能够正常拨上去，说明配置文件没有错。 但是在openwrt下的openvpn客户端，就无法配置成功并连接，大佬不忙的时候能测试下吗。：DDD。

[测试用证书](https://drive.google.com/file/d/1LMwBO-XznKLfkIxTCSufDjm1AtJSYWhE/view?usp=sharing) 
username  :  test  
password  :  test

minkaka, 2021-08-28:
这里是 一些参考 https://openvpn.net/community-resources/reference-manual-for-openvpn-2-4/

kiddin9, 2021-08-28:
在/etc/config/openvpn中对应的位置加上

option username 'test'
option password 'test'

比如

```
config openvpn 'test'
	option config '/etc/openvpn/test.ovpn'
	option username 'test'
	option password 'test'
	option enabled '1'
```

minkaka, 2021-08-28:
> 在/etc/config/openvpn中对应的位置加上

五体投地，分分钟连上了。

closed 2021-12-29 by kiddin9