kiddin9/Kwrt #463: github actions编译修改IP和替换网卡位置

a717445834, 2021-09-01:
请问如果使用github编译的话修改管理IP要如何操作?
是否是修改这个文件
OpenWrt_x86-r2s-r4s/devices/common/default-settings
[image]
我现在用的是m93p,每次刷进去之后wan口和lan口互换,请问如何通过配置调换网卡位置

kiddin9, 2021-09-01:
是的,

交换网口复制 https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/master/devices/nanopi-r4s/patches/lan_wan.patch 这个文件到devices/common/patches/目录下

a717445834, 2021-09-02:
谢谢大佬,马上去试试

a717445834, 2021-09-02:
> 是的,
> 
> 交换网口复制 https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/master/devices/nanopi-r4s/patches/lan_wan.patch 这个文件到devices/common/patches/目录下

谢谢大佬,我马上去试一下

a717445834, 2021-09-02:
> 是的,
> 
> 交换网口复制 https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/master/devices/nanopi-r4s/patches/lan_wan.patch 这个文件到devices/common/patches/目录下

您好,我按照这个操作修改文件后编译发现提示错误
[image]
[image]
我对比了一下openwrt的源代码,这个补丁包应该没有错,但是不知道为啥编译失败了
我修改了IP,编译成功,然后现在修改网卡位置,发现编译失败了
请问您知道是什么情况吗

kiddin9, 2021-09-02:
要一模一样复制那个文件,我那个文件有14行,你复制过去就变成12行了,

a717445834, 2021-09-02:
> 要一模一样复制那个文件,我那个文件有14行,你复制过去就变成12行了,

这么严谨的吗,好的我试一下
我还想再加一个eht2网卡的是不是这样配置?行数也要保持一致
  
--- a/package/base-files/files/etc/board.d/99-default_network
+++ b/package/base-files/files/etc/board.d/99-default_network
@@ -8,8 +8,8 @@ board_config_update
 
 json_is_a network object && exit 0
 
-ucidef_set_interface_lan 'eth0'
-[ -d /sys/class/net/eth1 ] && ucidef_set_interface_wan 'eth1'
+ucidef_set_interface_wan 'eth0'
+[ -d /sys/class/net/eth1 ] && ucidef_set_interface_lan 'eth1'
+[ -d /sys/class/net/eth1 ] && ucidef_set_interface_lan 'eth2'
 board_config_flush

closed 2021-12-29 by kiddin9