kiddin9/Kwrt #465: qBittorrent不能保存下载路径问题,不能保存配置,当前配置只在重启前有效,重启丢失

a717445834, 2021-09-02:
qBittorrent用luci-app修改下载路径
[image]
用WEBUI修改下载路径
[image]
只要一重启进程或者机器,打开WEBUI设置发现下载地址又被还原成/etc/qBittorrent/downloads/了
查看启动文件vim /etc/init.d/qbittorrent
发现有关于启动路径的设置
[image]
但是修改后重启还是无效,而且还会报错
[image]
不知道是哪里的问题

a717445834, 2021-09-02:
已解决
修改vim /etc/init.d/qbittorrent
通过多次的测试发现
[image]
重启服务的时候它会检查你是否存在配置,如果存在,那么就修改你配置文件的这两项,但是他修改的有问题,所以导致配置文件乱七八糟,我直接注释了,然后再输出一个666表示对自己的肯定

kiddin9, 2021-09-02:
测试了没遇到你说的这个情况,

kiddin9, 2021-09-02:
@kiddin9 下载路径以luci中设置的为准,

a717445834, 2021-09-02:
> @kiddin9 下载路径以luci中设置的为准,

我把QB的配置文件删除,然后通过luci保存并应用,他的下载路径就没问题,然后再通过luci修改下载路径,他就会出错了

closed 2021-12-29 by kiddin9