kiddin9/Kwrt #472: 固件挺好，优点很多，入坑jellyfin，发现没核显。。。。

jack987-cloud, 2021-09-12:
固件挺好，优点很多，入坑jellyfin，可惜没核显驱动，4415u的cpu

kiddin9, 2021-09-13:
默认是有添加核显的, https://op.supes.top/firmware/x86_64/x86_64_kernel.config 这个是kernel config文件,

jack987-cloud, 2021-09-14:
[image]

jack987-cloud, 2021-09-14:
官方原版是带核显驱动的，自己编译了好几次，勾选luci-app-diskman就编译不成功，单独勾选luci-app-dockerman倒是能编译成功，界面显示docker，就是没有容器，镜像等的选项

kiddin9, 2021-09-14:
你测试了官方原版的 /dev/dri 中有驱动文件吗?

jack987-cloud, 2021-09-15:
v21.02.0，确定有

jack987-cloud, 2021-09-15:
用官方v21.02.0源+官方.config+官方的软件包可以编译成功

kiddin9, 2021-09-15:
截个/dev/dri图看看, 我用的就是官方的v21.02.0

jack987-cloud, 2021-09-15:
[image]

kiddin9, 2021-09-15:
你这哪是官方的,这是esir的固件,

jack987-cloud, 2021-09-15:
我现在用是esir的高大全，我编译的v21.02.0没带上diskman和docker就没用，但是我测试了，是有核显的，和这个一样

jack987-cloud, 2021-09-15:
我没用虚拟机都是直接刷的，，各大版本都用过，后来就一直用你的版本，挺稳定的

jack987-cloud, 2021-09-15:
https://github.com/jack987-cloud/5/actions/workflows/1.yml 就这个，确实有，官方源+官方几个插件，没dockerman和diskman

jack987-cloud, 2021-09-15:
esir的功能很全，就是小毛病多，不能恢复配置是个大bug。。。相当于没备胎的车

kiddin9, 2021-09-15:
有测试过新固件吗,核显驱动8月份以后的固件才有.

jack987-cloud, 2021-09-15:
你的固件？昨天试了下，没有

jack987-cloud, 2021-09-15:
我用闲置电脑装了ubuntu也编译成了，就是编译不能带diskman，docker得重新安装，核显驱动有。
wget https://downloads.openwrt.org/releases/19.07.8/targets/x86/64/config.buildinfo -O .config 然后再make menuconfig修改配置，不用make kernel_menuconfig编译出来的就带驱动

closed 2021-09-15 by jack987-cloud