kiddin9/Kwrt #476: Openclash自动关闭

NeeSan888, 2021-09-16:
Openclash每隔数天会自动停止，手工重启后恢复正常，发生概率频繁。日志信息如下：
2021-09-16 06:26:39 守护程序：检测到转发规则顺序错误，重置插件防火墙规则...
2021-09-16 06:25:39 OpenClash 启动成功，请等待服务器上线！
2021-09-16 06:25:39 第十步: 添加计划任务,启动进程守护程序...
2021-09-16 06:25:38 第九步: 重启 Dnsmasq 程序...
2021-09-16 06:25:38 第八步: 设置防火墙规则...
2021-09-16 06:25:38 第七步: 设置控制面板...
2021-09-16 06:25:37 第六步: 等待主程序下载外部文件...
2021-09-16 06:25:34 第五步: 检查内核启动状态...
2021-09-16 06:25:31 未检测到特殊配置，调用 Dev 内核启动...
2021-09-16 06:25:31 第四步: 启动主程序...
2021-09-16 06:25:30 正在设置Fake-IP黑名单...
2021-09-16 06:25:28 第三步: 修改配置文件...
2021-09-16 06:25:28 第二步: 组件运行前检查...
2021-09-16 06:25:28 第一步: 获取配置...
2021-09-16 06:25:28 OpenClash 开始启动...
2021-09-16 06:25:27 第六步：删除 OpenClash 残留文件...
2021-09-16 06:25:27 第五步: 重启 Dnsmasq 程序...
2021-09-16 06:25:27 第四步: 关闭 Clash 主程序...
2021-09-16 06:25:27 第三步: 关闭 OpenClash 守护程序...
2021-09-16 06:25:26 第二步: 删除 OpenClash 防火墙规则...
2021-09-16 06:25:26 第一步: 备份当前策略组状态...
2021-09-16 06:25:26 OpenClash 开始关闭...
2021-09-16 04:06:32 警告：OpenClash 目前处于未启用状态，请从插件页面启动本插件，脚本退出...

NeeSan888, 2021-09-17:
每天凌晨4点自动停止。看了crontab，不知道第一行是什么任务：
63 4 * * * sh /etc/rc.d/S99opkg >/dev/null 2>&1
0 0 * * * /usr/share/openclash/openclash.sh
0 0 * * * /usr/share/openclash/openclash_rule.sh
0 0 * * * /usr/share/openclash/openclash_ipdb.sh
0 0 * * * /usr/share/openclash/openclash_chnroute.sh

kiddin9, 2021-09-17:
自动更新插件用的, 可以在 系统 固件设置 中关闭

NeeSan888, 2021-09-17:
> 自动更新插件用的, 可以在 系统 固件设置 中关闭

把计划任务调整了，应该没问题了。

closed 2021-09-17 by NeeSan888