kiddin9/Kwrt #479: x86安装9-13版本启动黑屏，第一次安装，是不是开机引导啥的要设置啥的？

chouxiang99, 2021-09-21:


kiddin9, 2021-09-21:
等一会,启动需要时间.

chouxiang99, 2021-09-21:
> 等一会,启动需要时间.

但是按理说应该是启动之后有显示吧  我启动之后  显示器直接是无信号

kiddin9, 2021-09-21:
其他固件正常么,

chouxiang99, 2021-09-21:
大概试了四个固件  有一个固件正常  然后剩下的三个固件都不行。。。刷机的软件有没有讲究？能不能推荐一下   或者有没有教程啥的？   我大概的操作就是u盘启动进入pe  然后使用dg删除软路由固态的分区  然后用软件写入固件，写入成功重启就是黑屏

chouxiang99, 2021-09-21:
> 其他固件正常么,

大概试了四个固件 有一个固件正常 然后剩下的三个固件都不行。。。刷机的软件有没有讲究？能不能推荐一下 或者有没有教程啥的？ 我大概的操作就是u盘启动进入pe 然后使用dg删除软路由固态的分区 然后用软件写入固件，写入成功重启就是黑屏

kiddin9, 2021-09-21:
那可能是你硬件的问题, 试试用非efi版 https://op.supes.top/firmware/x86_64/09.13-openwrt-x86-64-generic-squashfs-combined.img.gz

chouxiang99, 2021-09-21:
> 那可能是你硬件的问题, 试试用非efi版 https://op.supes.top/firmware/x86_64/09.13-openwrt-x86-64-generic-squashfs-combined.img.gz

好的  我去试试      感谢

chouxiang99, 2021-09-21:
> 那可能是你硬件的问题, 试试用非efi版 https://op.supes.top/firmware/x86_64/09.13-openwrt-x86-64-generic-squashfs-combined.img.gz

我的软路由  好像只支持legacy启动   是不是因为这个原因呀？

kiddin9, 2021-09-21:
https://op.supes.top/firmware/x86_64/09.13-openwrt-x86-64-generic-squashfs-combined.img.gz 这个就是支持legacy启动

chouxiang99, 2021-09-21:
> https://op.supes.top/firmware/x86_64/09.13-openwrt-x86-64-generic-squashfs-combined.img.gz 这个就是支持legacy启动

感谢大佬一直耐心的回我   我感觉我的软路由是不是主板被淘宝店家改过    那个淘宝店家给的固件    我随便刷都可以  但是我从恩山论坛找的四五个固件  都不可以  操作都是一摸一样的   但是结果就是只有淘宝店家给我的固件可以正常刷入并且重启

kiddin9, 2021-09-21:
你的该不会是x86的CPU吧, x86和x86_64不是通用的.

chouxiang99, 2021-09-21:
> 你的该不会是x86的CPU吧, x86和x86_64不是通用的.

那你这么一说，我的cpu是1037u    应该是x86的。。。。所以和这个x86_64是不兼容的是嘛。。。。

chouxiang99, 2021-09-21:
> 你的该不会是x86的CPU吧, x86和x86_64不是通用的.

我的是1037u   这个是x86还是x86_64呀  我查了一下   好像是支持64位的呢

chouxiang99, 2021-09-21:
> 你的该不会是x86的CPU吧, x86和x86_64不是通用的.

我找到原因了  不知道为啥  刷你这个固件   以及恩山的一些固件  刷机之后   显示器是黑屏的   但是我用xshell去连接  是可以链接上的  现在已经可以了

closed 2021-09-21 by chouxiang99