kiddin9/Kwrt #500: web端口更改

Duneon, 2021-10-30:
系统设置里面看不到Web管理啊，怎样修改？
我需要使用这个https://github.com/skydevil88/XboxDownload/blob/master/README_OpenWrt.md

kiddin9, 2021-10-30:
没有安装uhttpd,用的就是nginx, 所有不需要更改端口和安装nginx这一步,

Duneon, 2021-10-30:
nginx_init: NOT using conf file!
show config to be used by: nginx -T -c '/etc/nginx/uci.conf'

> 没有安装uhttpd,用的就是nginx, 所有不需要更改端口和安装nginx这一步,

我跳过那两个步骤SSH进路由编辑好xbox.conf后直接执行/etc/init.d/nginx restart命令报错

nginx_init: NOT using conf file!
show config to be used by: nginx -T -c '/etc/nginx/uci.conf'

kiddin9, 2021-10-30:
我测试了没问题, nginx -T -c '/etc/nginx/uci.conf 看报什么错,

Duneon, 2021-10-30:
> nginx_init: NOT using conf file! show config to be used by: nginx -T -c '/etc/nginx/uci.conf'
> 
> > 没有安装uhttpd,用的就是nginx, 所有不需要更改端口和安装nginx这一步,
> 
> 我跳过那两个步骤SSH进路由编辑好xbox.conf后直接执行/etc/init.d/nginx restart命令报错
> 
> nginx_init: NOT using conf file! show config to be used by: nginx -T -c '/etc/nginx/uci.conf'

Duneon, 2021-10-30:
> 我测试了没问题, nginx -T -c '/etc/nginx/uci.conf 看报什么错,

没有信息显示，只有一个 > 符号

kiddin9, 2021-10-30:
nginx -T -c '/etc/nginx/uci.conf'

Duneon, 2021-10-30:
> nginx -T -c '/etc/nginx/uci.conf'

 [emerg] 18128#0: could not build server_names_hash, you should increase server_names_hash_bucket_size: 32
nginx: configuration file /etc/nginx/uci.conf test failed

Duneon, 2021-10-30:
> nginx -T -c '/etc/nginx/uci.conf'

大佬还在看吗

Duneon, 2021-10-31:
解决了，代码上面加个server_names_hash_bucket_size 64
数字可以改，32倍数往上加，64  128 512
web端口还是要改8080

closed 2021-10-31 by Duneon

Duneon, 2022-01-10:
> 

https://bbs.a9vg.com/forum.php?mod=viewthread&tid=8781059&mobile=2

cwdsun, 2022-01-11:
> 
十分感谢，很有帮助。加了代码就成功了。
我没改8080端口，也能成功跳转

lan852, 2022-01-23:
> 

这个代码怎么加啊？   加在哪里呢