kiddin9/Kwrt #502: 10.26.2021 版搭配 adguard home 使用，重启后 bypass 无法正常启动

aQuaYi, 2021-10-31:
r2s 安装的 10.26.2021 版固件，只启用了 bypass 和 adguard home 插件。
adguard home 中只自定义了过滤规则，bypass 中勾选了与AdGuardHome搭配。
完成以上设置后，可以正常使用。
但是重启系统后，adguardhome可以正常启动，bypass 中 smartdns 与 chinadns ng 无法正常启动运行。导致电脑等设备无法访问外网。

目前，我的处理方式是，
1.在 adguardhome 中取消启用，
2.在 bypass 中取消勾选与adguardhome搭配，
3.重启r2s

kiddin9, 2021-10-31:
多重启几次试试,我这边没问题.

aQuaYi, 2021-10-31:
重启了好多遍了，这个问题一直都在，smartdns 与 chinadns 总是显示未运行。
我是设置了晚上12点12分自动重启的。如果不能重启就成功，要反复重启也太麻烦了。

补充信息，我在bypass中设置是
[image]

aQuaYi, 2021-10-31:
adguardhome的设置是
[image]

kiddin9, 2021-10-31:
如果取消了bypass中的 与AdGuardHome搭配 那就不要开启adguardhome了,
可以考虑使用 luci-app-dnsfilter 来去广告,

aQuaYi, 2021-10-31:
好的，我试试看dnsfilter。
不过，还是希望可以搭配bypass和adguardhome使用。。。

closed 2021-12-29 by kiddin9