kiddin9/Kwrt #503: opkg update 错误

xseruer, 2021-11-05:
Downloading https://op.supes.top/packages/aarch64_generic/base/Packages.gz
*** Failed to download the package list from https://op.supes.top/packages/aarch64_generic/base/Packages.gz

Downloading https://op.supes.top/packages/aarch64_generic/Packages.gz
*** Failed to download the package list from https://op.supes.top/packages/aarch64_generic/Packages.gz

Downloading https://op.supes.top/packages/aarch64_generic/luci/Packages.gz
*** Failed to download the package list from https://op.supes.top/packages/aarch64_generic/luci/Packages.gz

Downloading https://op.supes.top/packages/aarch64_generic/packages/Packages.gz
*** Failed to download the package list from https://op.supes.top/packages/aarch64_generic/packages/Packages.gz

Downloading https://op.supes.top/packages/aarch64_generic/routing/Packages.gz
*** Failed to download the package list from https://op.supes.top/packages/aarch64_generic/routing/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://op.supes.top/packages/aarch64_generic/base/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://op.supes.top/packages/aarch64_generic/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://op.supes.top/packages/aarch64_generic/luci/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://op.supes.top/packages/aarch64_generic/packages/Packages.gz, wget returned 255.
 * opkg_download: Failed to download https://op.supes.top/packages/aarch64_generic/routing/Packages.gz, wget returned 255.

---------------------------------------

估计版本：
OpenWrt 11.01.2021 by Kiddin' / LuCI openwrt-21.02 branch git-f28aaa3

kiddin9, 2021-11-05:
路由器网络没通.

rdcyuxi, 2021-12-08:
网络通的，但是更新busybox后，好像是wget没了
root@OpenWrt 13:25 ~# ping www.baidu.com
PING www.baidu.com (180.101.49.12): 56 data bytes
64 bytes from 180.101.49.12: seq=0 ttl=53 time=11.119 ms
64 bytes from 180.101.49.12: seq=1 ttl=53 time=11.521 ms
64 bytes from 180.101.49.12: seq=2 ttl=53 time=11.339 ms
64 bytes from 180.101.49.12: seq=3 ttl=53 time=11.255 ms
^C
--- www.baidu.com ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 11.119/11.308/11.521 ms
root@OpenWrt 13:25 ~# which wget
root@OpenWrt 13:25 ~# 
root@OpenWrt 13:28 ~# ll /usr/libexec/wget-ssl 
-rwxr-xr-x    1 root     root        521473 Oct 24 17:01 /usr/libexec/wget-ssl*
root@OpenWrt 13:28 ~# ll /bin/uclient-fetch 
-rwxr-xr-x    1 root     root         24595 Oct 24 17:01 /bin/uclient-fetch*
root@OpenWrt 13:28 ~#

kiddin9, 2021-12-08:
@rdcyuxi busybox是不能更新的,

closed 2021-12-29 by kiddin9