kiddin9/Kwrt #508: 固件作为旁路由网络该如何设置呢

madman43, 2021-11-11:
这是网络配置，我已经把默认桥接取消了，填入了ip但是还是ping不出去
[image]
[image]
[image]
[image]

还望能够指导下，感激不尽

kiddin9, 2021-11-11:
网关指向主路由, 设置DNS, 关闭DHCP,  其他都不用设置

madman43, 2021-11-11:
一直没用过新版openwrt，现在直接设置就好了，太方便了，感谢

closed 2021-11-11 by madman43

qifei, 2021-11-15:
此时旁路由的 LAN 口接主路由的 LAN 口对吗？，然后如何管理旁路由？
谢谢

madman43, 2021-11-15:
> 此时旁路由的 LAN 口接主路由的 LAN 口对吗？，然后如何管理旁路由？ 谢谢

旁路由的 LAN 口接主路由的 LAN 口，ip填同网段ip，关闭dhcp即可

qifei, 2021-11-15:
已经连上了，谢谢。并没有别人说的那么难。