kiddin9/Kwrt #512: R4S 的风扇控制脚本补丁的链接已经过期了，希望更新一下

macvip, 2021-11-17:
[diy.sh 里的这一行风扇控制脚本链接](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/454ac2e96b0a67bafebfd5aa86a95dbc14d9ebbd/devices/nanopi-r4s/diy.sh#L13)已经过期，结果编译出来的固件就启动不了 `fa-pwmfan` 这个控制风扇的服务了。
新的官方脚本比如 [master-21.02](https://github.com/friendlyarm/friendlywrt) 版本
链接地址在这里：
[/usr/bin/fa-fancontrol.sh](https://github.com/friendlyarm/friendlywrt/raw/master-v21.02/target/linux/rockchip/armv8/base-files/usr/bin/fa-fancontrol.sh)
[/usr/bin/fa-fancontrol-direct.sh](https://github.com/friendlyarm/friendlywrt/raw/master-v21.02/target/linux/rockchip/armv8/base-files/usr/bin/fa-fancontrol-direct.sh)
服务启动
[/etc/init.d/fa-fancontrol](https://github.com/friendlyarm/friendlywrt/raw/master-v21.02/target/linux/rockchip/armv8/base-files/etc/init.d/fa-fancontrol)

希望能把新的风扇控制脚本加一下，多谢🙏

kiddin9, 2021-11-17:
已修复.

macvip, 2021-11-17:
> 已修复.

修复的这里 https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/28c3fc64f31ef705ea28a0fbf0b09f920dc4ea4c/devices/nanopi-r4s/diy.sh#L16
`fa-fancontrol.sh` `fa-fancontrol-direct.sh` 这两个脚本都是以 **fa-** 开头，不是 fa**n**- 开头，没有字母 n
麻烦 @kiddin9  您再改一下。非常感谢。

kiddin9, 2021-11-17:
感谢指正, 已修复, 我想当然 风扇 那肯定是 fan 了...

macvip, 2021-11-17:
@kiddin9 太感谢啦

closed 2021-11-17 by macvip