kiddin9/Kwrt #517: 奇怪的现象，Booting the kernel之后

OUZYcn, 2021-11-24:
HW：PowerEdge R430
OS：DellEMC-VMware ESXi 7.0 Update 2 Build-18538813 (A07)
CPU：Intel(R) Xeon(R) CPU E5-2630L v4 @ 1.80GHz
NIC：NetXtreme BCM5720 Gigabit Ethernet 
NIC：QLogic 25GE 2P QL41262HxCU-DE Adapter
Rls：https://op.supes.top/firmware/x86_64/11.22-openwrt-x86-64-generic-squashfs-combined.img.gz

现象就是这样，进不到bash
[image]

这有点迷糊了，第一次见到这样的。。。
难道不成是上述两个NIC都没有驱动？ 但不应该进不到bash啊。。。

适用于网络适配器的 Qlogic Linux 驱动程序源代码：
https://www.dell.com/support/home/zh-cn/drivers/driversdetails?driverid=rtmtk

kiddin9, 2021-11-24:
确实有点bug,第一次开机要久等一会,如果一分钟后还是卡住,那就是网卡没驱动起来.

OUZYcn, 2021-11-24:
> 

那么这样很有可能两个NIC都没有驱动？

OUZYcn, 2021-11-24:
看来是驱动缺失问题居多，在Ubuntu server 20.10里边是OK的

```
ouzy@ouzy:~$ sudo lspci | grep Ethernet
03:00.0 Ethernet controller: VMware VMXNET3 Ethernet Controller (rev 01)
0b:00.0 Ethernet controller: QLogic Corp. FastLinQ QL41000 Series Gigabit Ethernet Controller (SR-IOV VF) (rev 02)
```

OUZYcn, 2021-11-24:
这个应该就是在Ubuntu中的驱动源代码，怎么加入到你的里边一起编译？

https://github.com/torvalds/linux/tree/master/drivers/net/ethernet/qlogic

kiddin9, 2021-11-24:
要弄成openwrt插件才行, 有试过其他人的op固件能正常驱动吗,

OUZYcn, 2021-11-24:
另外，**make kernelconfig** 是不行的，你改过？

```
make[1] kernelconfig
make -r kernelconfig: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/ouzy/openwrt/include/toplevel.mk:230: kernelconfig] Error 1
ouzy@ouzy:~/openwrt$ make kernelconfig V=sc
make[1]: Entering directory '/home/ouzy/openwrt'
make[1]: *** No rule to make target 'kernelconfig'.  Stop.
make[1]: Leaving directory '/home/ouzy/openwrt'
make: *** [/home/ouzy/openwrt/include/toplevel.mk:230: kernelconfig] Error 2
```

kiddin9, 2021-11-24:
先编译一次固件再 make kernelconfig

OUZYcn, 2021-11-24:
> 要弄成openwrt插件才行, 有试过其他人的op固件能正常驱动吗,

同样的虚拟机（仅有vmxnet虚拟网卡）配置，我替换磁盘文件为另外一个X86_64固件是可以的。。。。

Firmware Version：OpenWrt / LuCI openwrt-21.02 branch git-21.231.26241-422c175
Kernel Version ：5.4.143

```
root@OpenWrt:~# dmesg | grep vmxnet3
[    0.363004] VMware vmxnet3 virtual NIC driver - version 1.4.17.0-k-NAPI
[    0.363170] vmxnet3 0000:13:00.0: # of Tx queues : 2, # of Rx queues : 2
[    0.365004] vmxnet3 0000:13:00.0 eth0: NIC Link is Up 10000 Mbps
[    1.158935] vmxnet3 0000:13:00.0 eth0: intr type 3, mode 0, 3 vectors allocated
[    1.159953] vmxnet3 0000:13:00.0 eth0: NIC Link is Up 10000 Mbps
[    9.099157] vmxnet3 0000:13:00.0 eth0: intr type 3, mode 0, 3 vectors allocated
[    9.099893] vmxnet3 0000:13:00.0 eth0: NIC Link is Up 10000 Mbps
```

kiddin9, 2021-11-24:
重刷下 https://op.supes.top/firmware/x86_64/11.22-openwrt-x86-64-generic-squashfs-combined.img.gz 试试, 有修复了进不到 ash的问题,

OUZYcn, 2021-11-24:
> 重刷下 https://op.supes.top/firmware/x86_64/11.22-openwrt-x86-64-generic-squashfs-combined.img.gz 试试, 有修复了进不到 ash的问题,

等下告诉你情况

closed 2021-12-29 by kiddin9