kiddin9/Kwrt #520: RTL8811CU/RTL8821CU无线网卡驱动

djl0629, 2021-11-28:
网上不少用这个芯片的无线网卡搭配R2S，大大的固件插上无线网卡重启后可以看到无线选项多出来，可以扫描到附近的无线信号，但是设置AP模式后就变成设备未激活，切换回客户端模式又正常了，在AP模式下重启可以看到无线工作在AP模式下，但是其他设备搜索不到无线。我刚接上无线网卡的时候其实不是这样，刚接上的时候无论我怎么设置都会自动跳回客户端模式，然后我设置国家代码US，信道149，能够搜索到无线，但是无法分配ip，我认为可能是我网桥设置有问题，在折腾的过程中有一次LAN口连接的设备无法分配IP了，但是从无线连接的可以正常访问后台了，所以我认为其实无线可以被正确驱动，但是存在一些BUG。在折腾过程中我误操作导致无法进入后台，重新刷写固件后情况就和现在一样了，怎么也无法搜索到无线信号

djl0629, 2021-11-28:
我怀疑是无线网卡故障导致ap模式不可用，然后我现在把它插在win的pc上试了一下，是可以正常工作的，无论是打开移动热点发射信号还是作为客户端连接wifi。我之前确确实实连接了OpenWrt的wifi进入了管理后台，现在怎么设置都无法搜索到wifi信号

roxsky, 2021-12-22:
我也是这网卡，这网卡无法在web下设置的，我之前用另外个固件的时候都是在sh里面用脚本设置。
https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/usb-wifi.sh
就是这个脚本，你可以试试。

djl0629, 2021-12-22:
> 

谢谢 一会儿试试

roxsky, 2021-12-22:
> > 
> 
> 谢谢 一会儿试试

还有就是kiddin9大大的固件好像默认没有驱动，你有装驱动吗？装的是哪个？

djl0629, 2021-12-22:
> > > 
> > 
> > 
> > 谢谢 一会儿试试
> 
> 还有就是kiddin9大大的固件好像默认没有驱动，你有装驱动吗？装的是哪个？

默认是有驱动的，我自编译的时候也编译进去了，结果是一样的。对了这个脚本要怎么设置ac n legacy呢，生效之后能否再web里进行管理，信道是只能40吗

roxsky, 2021-12-22:
> 

我是用的每天自动编译的默认固件，自编译还不会，默认进去好像是没驱动的，我是安装了kmod-rtl8821cu这个包才识别出来无线网卡的，就算用了这个脚本也不能用。据我所知这个生效后也是不能改的，在openwrt下速率最高也只有54M，好像是驱动问题导致。

closed 2021-12-29 by kiddin9