kiddin9/Kwrt #532: R4S更换LAN口IP后重启，nginx进程无法启动

YUMEYA, 2021-12-06:
环境：R4S LAN、WAN口均配置了PPPOE拨号连接光猫，通过USB网卡连接AP模式的无线路由器，USB网卡配置的静态IP是10.1.1.1
现象：重启后SSH可以登录，Web界面提示访问被拒绝，查看进程nginx未启动，查看日志有nginx启动失败报配置错误的记录：
[image]
对应行的配置：
[image]
不加载shortcuts.conf时nginx可以正常启动

kiddin9, 2021-12-06:
lan使用USB网卡的缘故,导致获取lan口IP生成shortcuts.conf失败了,
删掉shortcuts.conf文件的话下回更新固件就又有了,直接把shortcuts.conf里的内容清空吧,

YUMEYA, 2021-12-07:
> lan使用USB网卡的缘故,导致获取lan口IP生成shortcuts.conf失败了, 删掉shortcuts.conf文件的话下回更新固件就又有了,直接把shortcuts.conf里的内容清空吧,

好的，谢谢大佬

closed 2021-12-07 by YUMEYA