kiddin9/Kwrt #534: Docker版不能连上外网，只有wan口没lan口

cryingcheems, 2021-12-08:
系统ubuntu20.04(x86_64)
按教程来的,咋设置都不行，自己修改/etc/config/network加lan口也不行，防火墙也试了，都不行，都连不上外网，内网倒是能ping通
求用过的指导下

kiddin9, 2021-12-08:
确实是镜像出了问题, 已经修复,请重新下载

cryingcheems, 2021-12-08:
好的，已成功，但是最新的镜像好像/etc/resolv.conf下的nameserver有点问题，文件中是127.0.0.11貌似多打了一个1，QAQ

cryingcheems, 2021-12-08:
> 确实是镜像出了问题, 已经修复,请重新下载

好的，已拉取最新镜像，有br-lan了，但是最新的镜像好像/etc/resolv.conf下的nameserver有点问题，文件中是127.0.0.11貌似多打了一个1，导致只能ping通ip，域名ping不通，QAQ

kiddin9, 2021-12-08:
确实多了个1,但是不影响上网,设置向导中 
[image]
亲测设置向导中设置好这三项就可以上网了,
[image]

cryingcheems, 2021-12-08:
十分感谢大佬，之前这个镜像折磨了我好久，终于解决了

closed 2021-12-08 by cryingcheems