kiddin9/Kwrt #537: $REPO_BRANCH ？

OUZYcn, 2021-12-13:
https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/9f1a4c25a75ae01139c4e1daeb6f7e79aad76443/onekey/build.sh#L91

虽然官方repo是备选，但是为啥 1. tentcent mirror上拿最新的rls，但是官方哪儿却拿snapshot呢？2. 官方都rls 21.02.1不需要21.02 snapshot吧？
我个人认为与前面 **$REPO_BRANCH** 保持一致 就好了，还是说有特别原因？

```
if [[ $firmware == "x86_64" ]]; then
	curl -fL -o sdk.tar.xz https://mirrors.cloud.tencent.com/openwrt/releases/$REPO_BRANCH/targets/x86/64/openwrt-sdk-$REPO_BRANCH-x86-64_gcc-8.4.0_musl.Linux-x86_64.tar.xz || curl -fL -o sdk.tar.xz https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/x86/64/openwrt-sdk-21.02-SNAPSHOT-x86-64_gcc-8.4.0_musl.Linux-x86_64.tar.xz
elif [[ $firmware == nanopi-* ]]; then
	curl -fL -o sdk.tar.xz https://mirrors.cloud.tencent.com/openwrt/releases/$REPO_BRANCH/targets/rockchip/armv8/openwrt-sdk-$REPO_BRANCH-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64.tar.xz || curl -fL -o sdk.tar.xz https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/rockchip/armv8/openwrt-sdk-21.02-SNAPSHOT-rockchip-armv8_gcc-8.4.0_musl.Linux-x86_64.tar.xz
elif [[ $firmware == "Rpi-4B" ]]; then
	curl -fL -o sdk.tar.xz https://mirrors.cloud.tencent.com/openwrt/releases/$REPO_BRANCH/targets/bcm27xx/bcm2711/openwrt-sdk-$REPO_BRANCH-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64.tar.xz || curl -fL -o sdk.tar.xz https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/bcm27xx/bcm2711/openwrt-sdk-21.02-SNAPSHOT-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64.tar.xz
fi
```

kiddin9, 2021-12-13:
因为官方最新的正式版release对应的SDK更新有延迟,一般刚发布的前几天是不会更新相应的sdk的.所以用snapshot做备用.

OUZYcn, 2021-12-13:
假定当执行build.sh的时点来看：
1.  $REPO_BRANCH 从官方github得到的21.02.1 release version 
https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/9f1a4c25a75ae01139c4e1daeb6f7e79aad76443/onekey/build.sh#L85

2.  然后去tencent mirror site去取得-SDK-**$REPO_BRANCH**-文件，“||” 若是失败就去官方取snapshot
https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/9f1a4c25a75ae01139c4e1daeb6f7e79aad76443/onekey/build.sh#L91

这时候存在你说的 “官方最新的正式版release对应的SDK更新有延迟” ？

PS，我在广东电信直连openwrt.org最快上10M+/s速度，反而tencent mirror 不见得最最快的。。。奇了

kiddin9, 2021-12-13:
假如官方今天发布了21.02.2, 那21.02.2对应的SDK一般要两三天后才会发布.

OUZYcn, 2021-12-13:
噢，对了顺便请教一下， 都拿的是最新的release了，这个diy.sh里边的一堆patch你都了解不？

https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/9f1a4c25a75ae01139c4e1daeb6f7e79aad76443/devices/common/diy.sh

```
svn co https://github.com/immortalwrt/immortalwrt/branches/openwrt-21.02/target/linux/generic/hack-5.4 target/linux/generic/hack-5.4
curl -sfL https://git.io/J0klM --create-dirs -o package/network/config/firewall/patches/fullconenat.patch
curl -sfL https://raw.githubusercontent.com/coolsnowwolf/lede/master/target/linux/generic/hack-5.4/601-netfilter-export-udp_get_timeouts-function.patch -o target/linux/generic/hack-5.4/601-netfilter-export-udp_get_timeouts-function.patch

```

kiddin9, 2021-12-13:
并不都了解.

OUZYcn, 2021-12-13:
后边两个能知道是啥，主要是这个太多了。。。
https://github.com/immortalwrt/immortalwrt/tree/openwrt-21.02/target/linux/generic/hack-5.4

`https://github.com/immortalwrt/immortalwrt/branches/openwrt-21.02/target/linux/generic/hack-5.4 target/linux/generic/hack-5.4`

OUZYcn, 2021-12-19:
这前头还有一些的，没能cp完全。 

错误应该是make download这里开始
make menuconfig 没怎么动过默认的.config，主要就是加入我自己要的Mellanox的驱动以及加入luci-app-openclash

这个dependency 错误之前是没有的。 
`WARNING: Makefile 'package/feeds/kiddin9/luci-app-turboacc/Makefile' has a dependency on 'dnsproxy', which does not exist
`
哦，对了，一闪而过看到dpdk，后来查了一下发现[openwrt-packages](https://github.com/kiddin9/openwrt-packages)  里边有 DPDK了。

```
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125357: No such file
cat: /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file or directory
/bin/sh: 3: cannot open /home/ouzy/kiddin9/openwrt/tmp/info/.files-packageinfo-1125459: No such file
WARNING: Makefile 'package/feeds/kiddin9/luci-app-rclone/Makefile' has a dependency on 'rclone-config', which does not exist
WARNING: Makefile 'package/feeds/kiddin9/luci-app-rclone/Makefile' has a dependency on 'rclone-webui-react', which does not exist
WARNING: Makefile 'package/feeds/kiddin9/luci-app-rclone/Makefile' has a dependency on 'rclone-ng', which does not exist
WARNING: Makefile 'package/feeds/kiddin9/luci-app-turboacc/Makefile' has a dependency on 'dnsproxy', which does not exist
 
```

closed 2021-12-29 by kiddin9