kiddin9/Kwrt #540: openclash更新固件之后，无法启动，也无法重新安装

Sean-2007, 2021-12-17:
Collected errors:
 * opkg_download: Failed to download https://op.supes.top/packages/x86_64/iptables-mod-extra_1.8.7-1_x86_64.ipk, wget returned 8.

stardd1, 2021-12-17:
软件包 - 配置OPKG - /etc/opkg/distfeeds.conf 下面的内容先备份好，替换
src/gz openwrt_core https://mirrors.aliyun.com/openwrt/releases/21.02.1/targets/x86/64/packages
src/gz openwrt_base https://mirrors.aliyun.com/openwrt/releases/21.02.1/packages/x86_64/base
src/gz openwrt_luci https://mirrors.aliyun.com/openwrt/releases/21.02.1/packages/x86_64/luci
src/gz openwrt_packages https://mirrors.aliyun.com/openwrt/releases/21.02.1/packages/x86_64/packages
src/gz openwrt_routing https://mirrors.aliyun.com/openwrt/releases/21.02.1/packages/x86_64/routing
src/gz openwrt_telephony https://mirrors.aliyun.com/openwrt/releases/21.02.1/packages/x86_64/telephony
保存   
刷新列表
搜索  iptables-mod-extra  安装
再替换回备份好的源，刷新 安装

L1cardo, 2021-12-17:
这样不行，会提示`has no valid architecture`

stardd1, 2021-12-17:
我刚刚就是这么干装上的，现在已经正常使用了

L1cardo, 2021-12-17:
R2S改成这个

```
src/gz openwrt_core mirrors.aliyun.com/openwrt/releases/21.02.1/targets/rockchip/armv8/packages
src/gz openwrt_base mirrors.aliyun.com/openwrt/releases/21.02.1/packages/aarch64_generic/base
src/gz openwrt_luci mirrors.aliyun.com/openwrt/releases/21.02.1/packages/aarch64_generic/luci
src/gz openwrt_packages mirrors.aliyun.com/openwrt/releases/21.02.1/packages/aarch64_generic/packages
src/gz openwrt_routing mirrors.aliyun.com/openwrt/releases/21.02.1/packages/aarch64_generic/routing
src/gz openwrt_telephony mirrors.aliyun.com/openwrt/releases/21.02.1/packages/aarch64_generic/telephony
```

Sean-2007, 2021-12-18:
> iptables-mod-extra

按你的方法也不行，
x86升级到最新的【 12.18.2021 】， iptables-mod-extra 安装，再替换回备份好的源，刷新 安装，提示

Collected errors:
 * pkg_run_script: package "luci-app-openclash" preinst script returned status 1.
 * preinst_configure: Aborting installation of luci-app-openclash.
 * opkg_install_cmd: Cannot install package luci-app-openclash.

closed 2021-12-29 by kiddin9