kiddin9/Kwrt #542: 怎么挂载SMB啊？

Omoinemie, 2021-12-20:
用了各种方法，网上也搜遍了，都没挂载成功，机器是R2S。
mount -t cifs和mount.cifs都不行，求指导！

Omoinemie, 2021-12-20:
那个luci-app-cifs也挂载不上去！

kiddin9, 2021-12-20:
luci-app-cifs-mount

Omoinemie, 2021-12-20:
是这个，挂不上去

Omoinemie, 2021-12-20:
[image]
用命令行是这个结果：
`root@OPENWRT 16:52 mnt# mount.cifs //192.168.31.3/Disk-1T/ /mnt/ -o user=*,pass=*,rw,vers=1.0
mount.cifs: bad UNC (、//192.168.31.3/Disk-1T/)`

Omoinemie, 2021-12-20:
我解决了，那个界面里的挂载点要用\而不能用/

hy817120, 2021-12-24:
先在挂载点挂载好 然后,这么设置就行了,这个固件,不需要去输入什么领命,增加什么或者删除什么,可以直接挂载的.我就是这样,前段时间把固件玩坏了,直接恢复了出厂,然后重新挂载的.
[image]

hy817120, 2021-12-24:
自带的就有smb,我用的就是自带了的,没下载任何东西,很简单的, 如果要配合aria2,记得以root用户运行就行了.

hy817120, 2021-12-24:
[image]
[image]
[image]

Omoinemie, 2021-12-24:
> 自带的就有,不用去乱改东西 先在挂载点挂载好然后设置就行了，这个不需要去输入什么命令。我就是这样，前段时间把固件玩坏了，直接恢复出厂，然后重新挂载的。 [image] [image] [image] [image]

我是路由器挂载smb，不是分享smb

daiwei43211, 2021-12-28:
我也遇到同样的问题啊！是否可以联系到你吗？我的微信：747376326@qq.com

hy817120, 2021-12-28:
> > 自带的，不用去乱改东西先在挂载好然后设置就行了，这个不需要去什么命令。挂载的。 [image] [image] [image] [image]
> 
> 我是主机挂载，不是分享

你是路由器扩容吗? 如果是扩容,感觉直接挂载在目录下不就行了吗?  不过没试过,我没扩容,

closed 2021-12-29 by kiddin9