kiddin9/Kwrt #549: 因为固件集成了smartdns-le，所以无法安装原版smartdns，也就是无法使用 luci-app-smartdns

moxuanyuan, 2021-12-28:
希望大神可以改进一下。。

kiddin9, 2021-12-28:
@moxuanyuan 早几个版本就已经去掉了smartdns-le了

moxuanyuan, 2021-12-29:
> @moxuanyuan 早几个版本就已经去掉了smartdns-le了

luci-app-smartdns 里一直提示 “SmartDNS - 未运行”，怎样解决？

kiddin9, 2021-12-29:
关闭bypass

moxuanyuan, 2021-12-29:
> 关闭bypass

意思是不能跟luci-app-smartdns 同时运行？我试了，关了bypass，确实可以，但只要一开bypass，就不行。。难度不能同时运行吗？

kiddin9, 2021-12-29:
bypass里已经集成了smartdns, 里面的smartdns是由bypass控制的, 当然与luci-app-smartdns冲突了,

moxuanyuan, 2021-12-29:
如果我想自定义smartdns 配置应该怎样搞？

moxuanyuan, 2021-12-29:
还有一个问题，在` 软件包 - 筛选器` 里输入 `smartdns`  在 `可用` 里点升级，是升级不了，是不是因为smartdns版本是内置的，跟软件仓库里的不一样，不能手动升级？

gxhunter1982, 2021-12-29:
> 还有一个问题，在` 软件包 - 筛选器` 里输入 `smartdns` 在 `可用` 里点升级，是升级不了，是不是因为smartdns版本是内置的，跟软件仓库里的不一样，不能手动升级？

可用的版本和已安装的不一样，都会在“可用”栏里显示“升级”选项。哪怕是可用库里的版本比已安装的旧，也会这么显示。
你应该去“升级”栏里看，有没有真正能升级的。

gxhunter1982, 2021-12-29:
> 还有一个问题，在` 软件包 - 筛选器` 里输入 `smartdns` 在 `可用` 里点升级，是升级不了，是不是因为smartdns版本是内置的，跟软件仓库里的不一样，不能手动升级？

今日最新版测试下来，已无这个问题

closed 2021-12-29 by kiddin9