kiddin9/Kwrt #574: opkg update报错，你的package包有些无法访问了，还请检查以下

zkh1992, 2022-01-23:
Collected errors:
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/packages/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/routing/Packages.gz, wget returned 8.
Collected errors:
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/packages/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/routing/Packages.gz, wget returned 8.

yingjun, 2022-01-24:
这是你的固件问题，你是不是把wget卸载掉了？你ssh到op,然后试试wget命令在不在。没有就装上，或者重置

sanfu2021, 2022-01-25:
请教一下，我也是这种现象，wget命令有，但我的是  returned 4，ssh 公匙 上那里找？

ssbbpp6162, 2024-12-03:
也遇到了这个问题 wget returned 8！请问怎么解决？ 谢谢!!