kiddin9/Kwrt #580: R2S更新后开始频繁断网

Q-V-O, 2022-02-01:
总是用一会儿就报WAN口错误, 只能重启恢复. 
[image]

kiddin9, 2022-02-01:
你是光猫拨号的吗?

Q-V-O, 2022-02-01:
> 你是光猫拨号的吗?

是的, 之前是用路由器拨号的, 之后换了宽带, 就用光猫拨号了,  因为线路问题现在也没办法用光猫拨号了, 而且没有超级密码. 光猫还是个只有网口的 天翼网关4.0 .

kiddin9, 2022-02-01:
上一个版本没问题么,

Q-V-O, 2022-02-01:
> 上一个版本没问题么,

之前版本没现在频繁, 但一天也会出现一两次的样子,  这个我印象里出现的时间是 , 更新里出现 可以按需安装驱动还是什么的.

Q-V-O, 2022-02-01:
> 上一个版本没问题么,

当时那个版本是直接 WAN口错误用不了, 第二天更新才解决

kiddin9, 2022-02-01:
https://bf.supes.top/?version=21.02.1&target=rockchip%2Farmv8&id=friendlyarm_nanopi-r2s 在这里定制个固件
[image]
kmod-usb-net-rtl8152-vendor 改成 kmod-usb-net-rtl8152 试试,

Q-V-O, 2022-02-03:
> https://bf.supes.top/?version=21.02.1&target=rockchip%2Farmv8&id=friendlyarm_nanopi-r2s 在这里定制个固件 [image] kmod-usb-net-rtl8152-vendor 改成 kmod-usb-net-rtl8152 试试,

非常感谢, 问题应该已经解决, 昨天用了一天, 没有再出现接口错误 导致断网的问题.