kiddin9/Kwrt #583: 2.6 版本aria2 403提示账户没有权限访问下载的目录

madman43, 2022-02-08:
Tue Feb  8 22:57:58 2022 daemon.err nginx[2956]: 2022/02/08 22:57:58 [error] 2995#0: *396 directory index of "/www/ariang/" is forbidden, client: 192.168.2.20, server: op, request: "GET /ariang/ HTTP/1.1", host: "192.168.2.2", referrer: "http://192.168.2.2/luci/admin/services/aria2"
Tue Feb  8 22:57:58 2022 daemon.err nginx[2956]: 2022/02/08 22:57:58 [error] 2995#0: *396 open() "/www/favicon.ico" failed (2: No such file or directory), client: 192.168.2.20, server: op, request: "GET /favicon.ico HTTP/1.1", host: "192.168.2.2", referrer: "http://192.168.2.2/ariang/"
Tue Feb  8 22:58:26 2022 daemon.info aria2: Aria2 will run with user 'aria2'.
Tue Feb  8 22:58:26 2022 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sdc1

kiddin9, 2022-02-08:
安装 ariang 而不是 ariang-nginx 你是web打不开还是aria2不运行?

madman43, 2022-02-08:
> 安装 ariang 而不是 ariang-nginx 你是web打不开还是aria2不运行?

我是安装的aria2 不是ng也不是nginx，事运行后打不开web页面，第一次全新安装后是正常打开的，但填写key后直接弹403了

kiddin9, 2022-02-08:
不知道你说的是什么key, 把 /etc/config/aria2删掉后重新安装luci-app-aria2

madman43, 2022-02-08:
> 不知道你说的是什么key, 把 /etc/config/aria2删掉后重新安装luci-app-aria2

就是密钥，那我明天试试

madman43, 2022-02-09:
删除配置文件后重装还是403web页面无法访问

madman43, 2022-02-09:
我重新使用dd命令全新安装aria2还是报403

louis-zhou, 2022-02-09:
你试试访问 http://ip/ariang/index.html 看看。

默认点击界面访问的是 http://ip/ariang 缺了 /index.html 。

估计原因是 Ng 没配置默认 index 。

madman43, 2022-02-10:
> 你试试访问 http://ip/ariang/index.html 看看。
> 
> 默认点击界面访问的是 http://ip/ariang 缺了 /index.html 。
> 
> 估计原因是 Ng 没配置默认 index 。

进去了，但是输入rpc秘钥依旧是未连接，之前通过低版本安装后升级最新版本打开了也是链接上的，下载了两个东西今早起来看就又成那样了，用你的办法进去后无法连接😅

madman43, 2022-02-11:
> 

嗯嗯

madman43, 2022-02-14:
> > 你试试访问 http://ip/ariang/index.html 看看。
> > 默认点击界面访问的是 http://ip/ariang 缺了 /index.html 。
> > 估计原因是 Ng 没配置默认 index 。
> 
> 确实有这个问题, /etc/config/nginx 中的ng_index.html 后面加上 index.html 然后service nginx restart 可以解决.

额又有新问题，http模式下输入密码跳转到登录界面。https输入密码正常进入。
这是 /etc/config/nginx 配置信息

`
config main 'global'
	option uci_enable 'true'

config server '_lan'
	list listen '80 default_server'
	list listen '[::]:80 default_server'
	option server_name 'op'
	list include 'conf.d/*.locations'
	option access_log 'off; # logd openwrt'
	option index 'luci index.html index.htm'

config server '_ssl'
	list listen '443 ssl'
	list listen '[::]:443 ssl'
	option server_name '_ssl'
	list include 'conf.d/*.locations'
	option ssl_certificate '/etc/nginx/conf.d/_lan.crt'
	option ssl_certificate_key '/etc/nginx/conf.d/_lan.key'
	option ssl_session_cache 'shared:SSL:32k'
	option ssl_session_timeout '64m'
	option access_log 'off; # logd openwrt'
	option index 'luci index.html index.htm'

config server '_redirect2ssl'

`

closed 2022-02-21 by madman43