kiddin9/Kwrt #589: 现在编译一次好大啊。。。。

yhl452493373, 2022-02-14:
以前只有指定的固件，现在还有其他格式的固件和编译后的软件库，科学上网压力很大啊

kiddin9, 2022-02-14:
你把  https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/714aa96710158eb4b0eb4421d8bb59cd593e3a6e/devices/common/.config#L10  这行去掉, 编译速度就快了.

yhl452493373, 2022-02-15:
> @yhl452493373 不建议自己编译了,用 bf.supes.top 生成吧,

主要是我要单独加一些设置。比如把防火墙打开，替换部分菜单文字，交换lan1和lan0的顺序等。我之前试过用生成，只能定制安装的软件，需要单独增加的设置、增加或替换的文件没法上传。

yhl452493373, 2022-02-15:
> 你把
> 
> https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/714aa96710158eb4b0eb4421d8bb59cd593e3a6e/devices/common/.config#L10
> 
> 这行去掉, 编译速度就快了.

感谢，我试试

yhl452493373, 2022-02-15:
> @yhl452493373 不建议自己编译了,用 bf.supes.top 生成吧,

另外现在貌似还有个问题，我看nginx默认都转https了，但是会导致ttyd终端无法使用

yhl452493373, 2022-02-15:
> 你把
> 
> https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/714aa96710158eb4b0eb4421d8bb59cd593e3a6e/devices/common/.config#L10
> 
> 这行去掉, 编译速度就快了.

啊，我不是说速度慢，是编译后的包很大，接近300兆。原因是生成了combined.img.gz和rootfs.tar.gz，但是我只需要combined-efi.img.gz。以前是可以只生成combined-efi.img.gz的，现在貌似没有地方改这个设置了

kiddin9, 2022-02-15:
[CONFIG_TARGET_ROOTFS_TARGZ=y](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/96d41c897808e31f8a79df2684fb0bfa9a9f693b/devices/common/.config#L4) 这行可去掉 rootfs.tar.gz
combined.img.gz 没法去掉,你可以在 Organize files 这步里不包含combined.img.gz即可.

yhl452493373, 2022-02-15:
> [CONFIG_TARGET_ROOTFS_TARGZ=y](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s/blob/96d41c897808e31f8a79df2684fb0bfa9a9f693b/devices/common/.config#L4) 删掉这行可去掉 rootfs.tar.gz combined.img.gz 没法去掉,你可以在 Organize files 这步里不包含combined.img.gz即可.

好的，我试试。多谢