kiddin9/Kwrt #590: NFS目录无法访问

seanria, 2022-02-16:
我在软件包里安装了 luci-app-nfs，使用配置

root@OPENWRT 09:08 PM sda1# cat /etc/exports 
/mnt/sda1	*(rw,sync,root_squash,all_squash,insecure,no_subtree_check)

rpcbind和rpcd也已经启动
我在 Mac 上用 showmount 命令可以找到目录：
Exports list on 192.168.50.210:
/mnt/sda1                           *

但是挂载的时候就会提示
mount -t nfs 192.168.50.210:/mnt/sda1  ~/directory
nfs server 192.168.50.210:/mnt/sda1: not responding

请问是什么原因？是不是防火墙的问题？

seanria, 2022-02-16:
或者 ksmbd 该怎么配置呢？感觉怎么配置都无法访问到目录。求解答一下，弄了好久都没弄好，谢谢了

kiddin9, 2022-02-17:
允许匿名访问和强制root勾上,

seanria, 2022-02-17:
root@OPENWRT 08:13 PM /# /etc/init.d/ksmbd status
inactive

我的 ksmbd 一直是 inactive 的状态，start也没用，容器重建了很多次都是这样，是哪里没设置对吗？

madman43, 2022-02-20:
我也遇见问题。
[image]
不管新建账户还是强制root一直提示密码错误

bauw2008, 2022-02-21:
ksmbd 新版本密码错误，用旧版本就没问题，其中也有些版本不稳定。

kiddin9, 2022-02-21:
ksmbd 新版本确实有问题, ksmbd 总是大大小小一些毛病, 新固件已经换成 samba4了, 一会不会再使用ksmbd.