kiddin9/Kwrt #594: homebridge无法启动

ZhibinLiIntel, 2022-02-20:
如图，即使勾选了启用并保存应用，homebridge依然是未运行状态，这个插件是不是一直都是不能用的状态呢？
[image]

kiddin9, 2022-02-20:
没用过这插件, 你可以在这 https://github.com/shanglanxin/luci-app-homebridge 了解下

ZhibinLiIntel, 2022-02-22:
> 没用过这插件, 你可以在这 https://github.com/shanglanxin/luci-app-homebridge 了解下

这个插件就是集成在opkg源里的那个对吧？尝试了这个在网页的管理界面没办法启动，ssh到路由器后命令行启动，homekit也无法添加bridge，并且官方的homebridge用node安装也后也一样遇到homekit无法添加bridge的问题。

https://github.com/nxhack/openwrt-node-packages 这个repo下的node-homebridge可以正常使用，能正常启动并且IOS的home app可以添加该bridge，搜了一下这个`node-homebridge`在opkg的软件包里也能搜到。

closed 2022-02-22 by ZhibinLiIntel

kiddin9, 2022-02-22:
https://openwrt.org/packages/pkgdata/node-homebridge 是官方库中的.

ZhibinLiIntel, 2022-02-22:
> https://openwrt.org/packages/pkgdata/node-homebridge 是官方库中的.

谢谢回复。补充一下这个`node-homebridge`没有图形界面，无法在网页管理界面中对应选项，只能从命令行启动然后挂后台。