kiddin9/Kwrt #600: [R2S] [2022-02-22版本] bypass启用“与AdGuardHome搭配使用”导致无法分配ipv4地址

dxglaw, 2022-02-27:
非常感谢分享如此优秀的固件。

使用中遇到了两个问题，都与bypass相关，因此就在一个issue中都包含了。
设备是R2S，刚开始用2022-02-22版本。
1.问题1描述：
    1.1 在bypass里面启用“与AdGuardHome搭配使用”后，当时一切正常，可以看到AdGuardHome中dns请求数量在增加；
    1.2 一段时间后（第二天起床），设备无法联网，发现设备只获取了ipv6的地址，而没有ipv4地址；
    1.3 怀疑是dhcp租期过期后，dhcp没有分配ip地址；
    1.4 重启设备后，正常，但过一天后再次出现该问题。
 2. 问题2描述：
    2.1 在bypass里面启用“与AdGuardHome搭配使用”后，接口LAN里的dhcp option恢复默认设置。在启用AGH前，dhcp设置里面有两个option，用于指定AC地址
[image]
[image]
    2.2 默认设置小时后，可以再次添加。

kiddin9, 2022-02-27:
第一个问题你可以尝试将租约时间尽可能设置大些,家庭网络设备少不会有什么影响
第二个已修复,等待更新bypass即可.

dxglaw, 2022-02-28:
谢谢。祝好。

closed 2022-02-28 by dxglaw