kiddin9/Kwrt #612: 这个openwrt为啥有梦想星城的流量啊？

wsppxcn, 2022-03-07:
### 环境
1.爱快一个
2.openwrt一个
3.win10主机一个
都是作为单独一个主机存在，openwrt总是有梦想星城的流量和网易游戏更新的流量，每天都有上传流量！

[image]
[image]
[image]

kiddin9, 2022-03-07:
换固件.

closed 2022-03-07 by kiddin9

wsppxcn, 2022-03-07:
> 

升级了确实没有了，原来的固件是有这种上传的程序吗

wsppxcn, 2022-03-07:
这种流量更多了。。。 看样子得用别的固件了。

[image]

kiruya1987, 2022-04-24:
@wsppxcn 想问下这个监控插件叫什么?  谢谢