kiddin9/Kwrt #640: 3.28版本反复重启

madman43, 2022-04-02:
使用openclash路由器就反复重启。和在没启动clash的情况下点击保存会造成路由断网。只能重启解决。

kiddin9, 2022-04-02:
@madman43 卸载kmod-inet-diag试下,可能是和我今天临时添加了kmod-inet-diag导致的,

madman43, 2022-04-03:
> @madman43 卸载kmod-inet-diag试下,可能是和我今天临时添加了kmod-inet-diag导致的,
早期3.28也会

madman43, 2022-04-03:
Sun Apr  3 17:27:17 2022 daemon.info dnsmasq[24536]: Connected to system UBus
Sun Apr  3 17:27:53 2022 daemon.info dnsmasq[25706]: Connected to system UBus
Sun Apr  3 17:28:29 2022 daemon.info dnsmasq[26496]: Connected to system UBus
Sun Apr  3 17:29:05 2022 daemon.info dnsmasq[27291]: Connected to system UBus
Sun Apr  3 17:29:41 2022 daemon.info dnsmasq[28083]: Connected to system UBus
Sun Apr  3 17:30:16 2022 daemon.info dnsmasq[28875]: Connected to system UBus
Sun Apr  3 17:30:52 2022 daemon.info dnsmasq[29666]: Connected to system UBus
Sun Apr  3 17:31:28 2022 daemon.info dnsmasq[30456]: Connected to system UBus
Sun Apr  3 17:32:04 2022 daemon.info dnsmasq[31241]: Connected to system UBus
Sun Apr  3 17:32:40 2022 daemon.info dnsmasq[32026]: Connected to system UBus
Sun Apr  3 17:33:16 2022 daemon.info dnsmasq[342]: Connected to system UBus
Sun Apr  3 17:33:44 2022 daemon.info uwsgi-luci: Sun Apr  3 17:33:44 2022 - received message 0 from emperor
Sun Apr  3 17:33:52 2022 daemon.info dnsmasq[1168]: Connected to system UBus
Sun Apr  3 17:34:28 2022 daemon.info dnsmasq[1969]: Connected to system UBus
Sun Apr  3 17:35:04 2022 daemon.info dnsmasq[2796]: Connected to system UBus
Sun Apr  3 17:35:40 2022 daemon.info dnsmasq[3593]: Connected to system UBus
Sun Apr  3 17:36:16 2022 daemon.info dnsmasq[4379]: Connected to system UBus
Sun Apr  3 17:36:31 2022 daemon.info uwsgi-luci: spawned uWSGI master process (pid: 4478)
Sun Apr  3 17:36:52 2022 daemon.info dnsmasq[5575]: Connected to system UBus
Sun Apr  3 17:37:28 2022 daemon.info dnsmasq[6359]: Connected to system UBus
Sun Apr  3 17:38:04 2022 daemon.info dnsmasq[7149]: Connected to system UBus
Sun Apr  3 17:38:40 2022 daemon.info dnsmasq[7932]: Connected to system UBus
Sun Apr  3 17:39:16 2022 daemon.info dnsmasq[9009]: Connected to system UBus
Sun Apr  3 17:39:52 2022 daemon.info dnsmasq[9804]: Connected to system UBus
Sun Apr  3 17:40:28 2022 daemon.info dnsmasq[10587]: Connected to system UBus
Sun Apr  3 17:41:04 2022 daemon.info dnsmasq[11379]: Connected to system UBus
Sun Apr  3 17:41:40 2022 daemon.info dnsmasq[12163]: Connected to system UBus
Sun Apr  3 17:42:16 2022 daemon.info dnsmasq[12947]: Connected to system UBus
Sun Apr  3 17:42:52 2022 daemon.info dnsmasq[13736]: Connected to system UBus
Sun Apr  3 17:43:28 2022 daemon.info dnsmasq[14526]: Connected to system UBus
Sun Apr  3 17:44:03 2022 daemon.info dnsmasq[15314]: Connected to system UBus
Sun Apr  3 17:44:39 2022 daemon.info dnsmasq[16098]: Connected to system UBus
Sun Apr  3 17:44:51 2022 daemon.info uwsgi-luci: Sun Apr  3 17:44:51 2022 - received message 0 from emperor
Sun Apr  3 17:45:15 2022 daemon.info dnsmasq[16882]: Connected to system UBus
Sun Apr  3 17:45:51 2022 daemon.info dnsmasq[17665]: Connected to system UBus
Sun Apr  3 17:50:41 2022 daemon.info dnsmasq[19290]: Connected to system UBus
Sun Apr  3 17:56:37 2022 user.err ddns-scripts[3221]: myddns_ipv4: GNU Wget Error: '4'
Sun Apr  3 17:56:37 2022 user.warn ddns-scripts[3221]: myddns_ipv4: Transfer failed - retry 1/0 in 60 seconds
Sun Apr  3 17:57:59 2022 daemon.info dnsmasq[21363]: Connected to system UBus
Sun Apr  3 18:32:59 2022 daemon.info dnsmasq[28472]: Connected to system UBus
Sun Apr  3 18:36:19 2022 daemon.info dnsmasq[29733]: Connected to system UBus
Sun Apr  3 19:06:31 2022 daemon.info dnsmasq[3593]: Connected to system UBus
Sun Apr  3 19:19:30 2022 daemon.info dnsmasq[6724]: Connected to system UBus
Sun Apr  3 19:29:03 2022 daemon.info uwsgi-luci: spawned uWSGI master process (pid: 8583)
Sun Apr  3 19:29:06 2022 user.info : luci: accepted login on / for root from 192.168.2.20

路由器一直提示这个，反复唤醒电脑

closed 2022-04-14 by madman43