kiddin9/Kwrt #645: 又一飞跃，终于有i915驱动支持，核显问题反馈

genesis-lua, 2022-04-09:
发现最新的4月8日固件带了i915驱动，理应是可以支持核显硬解了。但是进入文件查看没发现i915.ko这个驱动，然后再去dri目录也没有发现renderD128转码模块，大佬有空看看是什么问题？CPU是i5-7500，多谢

kiddin9, 2022-04-09:
i915驱动确实还有问题,后续我修复下.

是内网测速上行下行都正常, 但是实际跑宽带测速,下行有问题吗?

kiddin9, 2022-04-09:
i915驱动已修复.

genesis-lua, 2022-04-09:
> i915驱动确实还有问题,后续我修复下.
> 
> 是内网测速上行下行都正常, 但是实际跑宽带测速,下行有问题吗?

目前测试是内网正常，宽带下行10g SFP+光口几乎为零，换网口下行350-500兆浮动，我换固件再测试下

genesis-lua, 2022-04-09:
> i915驱动已修复.

最新版本测试已经生成了i915.ko驱动文件，但未发现生成/dev/dri/目录，也没发现renderD128转码模块，麻烦大佬再抽空看看
ls: /dev/dri/: No such file or directory

btw:大佬用的是什么CPU测试？

genesis-lua, 2022-04-09:
更新测试情况：使用最新4月9日的固件宽带测速依然是无法跑满千兆（原生固件，不带多余软件），切换回3月28日版本固件光口和网口均能跑满千兆带宽（原生固件）

kiddin9, 2022-04-09:
@genesis-lua 你的网卡是什么型号. 跑带宽的时候,CPU占用怎么样.

zzk1996, 2022-04-10:
> i915驱动已修复.

牛逼大佬，给大佬递烟

genesis-lua, 2022-04-10:
> @genesis-lua 你的网卡是什么型号. 跑带宽的时候,CPU占用怎么样.

genesis-lua, 2022-04-10:
@kiddin9 我的网卡型号是mellanox MCX312A-XCBT，宽带测试的CPU占用截图：
[image]
[image]

genesis-lua, 2022-04-10:
@zzk1996 你的核显能硬解了吗？是什么CPU？

genesis-lua, 2022-04-10:
@kiddin9 用lean大的源码编译了个固件，那个解码模块可以出来
[image]

zzk1996, 2022-04-11:
> @zzk1996 你的核显能硬解了吗？是什么CPU？

我机器还没回来，还没测试过，i3-8100

genesis-lua, 2022-04-12:
> > @zzk1996 你的核显能硬解了吗？是什么CPU？
> 
> 我机器还没回来，还没测试过，i3-8100

回来了测试结果共享下呗？你有用万兆卡不？

zzk1996, 2022-04-12:
> 你

好的，不过我没万兆卡

zzk1996, 2022-04-12:
> > > @zzk1996 你的核显能硬解了吗？是什么CPU？
> > 
> > 我机器还没回来，还没测试过，i3-8100
> 
> 回来了测试结果共享下呗？你有用万兆卡不？

没有万兆卡

zzk1996, 2022-04-21:
> > > @zzk1996 你的核显能硬解了吗？是什么CPU？
> > 
> > 我机器还没回来，还没测试过，i3-8100
> 
> 回来了测试结果共享下呗？你有用万兆卡不？

回来了，出不来核显，等大哥修复吧
宝哥的iStoreOS也出了，你可以试试，有驱动

genesis-lua, 2022-04-21:
> 

有看到那个固件了，界面还行，插件少了点

zzk1996, 2022-04-21:
> > 
> 
> 有看到那个固件了，界面还行，插件少了点

试过了，装插件有点难受，老是报错，说架构不匹配

genesis-lua, 2022-05-01:
大佬，复测发现核显问题和万兆速率问题还是存在

kiddin9, 2022-05-02:
i915驱动已修复.更新至05.22版即可.

genesis-lua, 2022-05-07:
> 网卡型号是mellanox MCX312A-XCBT

显卡驱动测试很正常，但万兆网卡依然下行没有速度，网卡型号是mellanox MCX312A-XCBT，估计是内核和驱动的问题了