kiddin9/Kwrt #659: 重启后，固件的dns设置的部分选项被重置，主要是 “禁止解析 IPv6 DNS 记录”和“DNS 转发”的设置被重置，请问改如何处理

pzqpeter, 2022-04-17:


pzqpeter, 2022-04-17:
Mon Apr 18 01:30:34 2022 kern.notice kernel: [    0.000000] Linux version 5.15.31 (runner@fv-az183-695) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r16495-bf0c965af0) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Sat Apr 9 16:08:19 2022
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=2b503caf-cb5f-c9c5-5ac0-1a72959a0d02 rootwait nomodeset console=ttyS0,115200n8 noinitrd
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] signal: max sigframe size: 1104
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-provided physical RAM map:
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000760c0fff] usable
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000760c1000-0x00000000785c0fff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000785c1000-0x0000000078840fff] ACPI data
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000078841000-0x0000000078940fff] ACPI NVS
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000078941000-0x0000000078efefff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000078eff000-0x0000000078ffefff] type 20
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000078fff000-0x0000000078ffffff] usable
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000079000000-0x000000007fbfffff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002803fffff] usable
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] NX (Execute Disable) protection: active
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] efi: EFI v2.70 by American Megatrends
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] efi: ACPI=0x7891b000 ACPI 2.0=0x7891b014 SMBIOS=0x78d7b000 SMBIOS 3.0=0x78d7a000 MEMATTR=0x72d85018 ESRT=0x74f26518
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] SMBIOS 3.3.0 present.
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] DMI: Default string Default string/CW-N6000, BIOS 5.19 03/30/2022
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] tsc: Detected 2000.000 MHz processor
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000000] tsc: Detected 1996.800 MHz TSC
Mon Apr 18 01:30:34 2022 kern.debug kernel: [    0.000242] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
Mon Apr 18 01:30:34 2022 kern.debug kernel: [    0.000246] e820: remove [mem 0x000a0000-0x000fffff] usable
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000252] last_pfn = 0x280400 max_arch_pfn = 0x400000000
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000292] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.000592] last_pfn = 0x79000 max_arch_pfn = 0x400000000
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.026276] esrt: Reserving ESRT space from 0x0000000074f26518 to 0x0000000074f26550.
Mon Apr 18 01:30:34 2022 kern.debug kernel: [    0.026282] e820: update [mem 0x74f26000-0x74f26fff] usable ==> reserved
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.026771] Secure boot could not be determined
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.026773] ACPI: Early table checksum verification disabled
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.026776] ACPI: RSDP 0x000000007891B014 000024 (v02 ALASKA)
Mon Apr 18 01:30:34 2022 kern.info kernel: [    0.026780] ACPI: XSDT 0x000000007891A728 0000D4 (v01 ALASKA A M I    01072009 AMI  01000013)
[... cut ...]
Mon Apr 18 01:35:03 2022 daemon.err smbd[30490]: [2022/04/18 01:35:03.561191,  0] ../../lib/util/become_daemon.c:136(daemon_ready)
Mon Apr 18 01:35:03 2022 daemon.err smbd[30490]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Mon Apr 18 01:35:33 2022 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Mon Apr 18 01:35:33 2022 daemon.notice ksmbd: Starting Ksmbd userspace service.
Mon Apr 18 01:35:33 2022 kern.err kernel: [  307.512334] ksmbd: Failed to bind socket: -98
Mon Apr 18 01:35:33 2022 kern.err kernel: [  307.512828] ksmbd: Failed to shutdown socket: -107

kiddin9, 2022-04-17:
肯定是你安装的其他某个插件修改的,

pzqpeter, 2022-04-18:
补充说明，经过查找经常修改配置的插件是MosDNS，及时关闭自启动还是会有影响，有时卸载也不太好用，建议直接重新编译固件

kiddin9, 2022-04-18:
> 补充说明，经过查找经常修改配置的插件是MosDNS，及时关闭自启动还是会有影响，有时卸载也不太好用，建议直接重新编译固件

好的,  这个插件我没用过, 你可以去找这个插件的作者去反馈.

closed 2022-09-11 by pzqpeter