kiddin9/Kwrt #743: R2S

tmacwk, 2022-06-12:
您好，首先非常感谢你们做出了如此方便的工具和平台。为爱发电了终生。

今天构建好固件后，刷上了，非常好用，当我看到有个在线更新的地方，点了一下，说没有发现新固件，重刷当前固件，我不小心点了，点之前我确认上面两个勾 保存当前设置和保存现有软件插件，是勾上的，但是冲刷当前固件之后，什么都没了。

然后我重新刷卡，刷固件，奇怪的事情发生了，里面一些页面变成了英文界面，再插卡重刷依然如此，目前只能通过改变主题来解决，不知道什么原因。

最后，这个保存当前设置是不是无效啊。

kiddin9, 2022-06-12:
如果是定制固件的话,要通过系统->定制系统升级 来更新固件.  在线更新固件适合默认固件,会记录你在软件包中安装的插件,在线更新完成后再自动从云仓库中下载以前的固件, 如果网络不正常可能导致插件自动安装失败. 但是配置都不会丢的.

英文界面试试 ctrl+F5, 如果还不行, 试试先卡刷这个 https://op.supes.top/firmware/nanopi-r2s/Restore-SDCard-2G.img.zip ,然后再卡刷openwrt固件.

tmacwk, 2022-06-12:
哦，原来是在这样，明白了。英文的问题解决了，太好了。
最后一个事情，在可选包输入的时候发现这个，这两个adguardhome是有什么不同吗
[image]

kiddin9, 2022-06-12:
选择第二个, luci-app开头的是管理界面, adguardhome是核心程序, 安装管理界面就会自动安装对应的核心程序.

tmacwk, 2022-06-12:
收到！非常感谢！你们这个简直太棒了，太方便了。