kiddin9/Kwrt #744: R2S 上行带宽问题

gee1k, 2022-06-13:
你好，
目前 R2S 固件在不交换网口的情况下上传速度会被限制在10M左右，代理只有5M，下载速度没有问题。交换网口过后上传下载均恢复正常（100M）。
前几天在TG反馈过这个问题，刚刚看到有新版本固件。更新过后发现问题还在，特地来留给 Issue，等你有空的时候可以看看是什么问题！

然后顺带提一个另外的问题，刚刚在系统内部在线更新的时候选择了保存配置和保留插件，但是更新过后插件并没有自动安装恢复。特地等了很久，而且 opkg 并没有在占用。后自己手动重新安装插件，大部分配置倒的确还在（AdGuardHome的配置没了）

> 直连测速

[image]

> 代理测速

[image]

gee1k, 2022-06-13:
经过多次测试发现：直连情况下上传速度也不是完全是 10M 左右，有时候会出现测速刚开始时上传速度跑到 100M 左右，然后在一两秒钟之内急速下降到50M以下。

代理情况下多次测试一直是5M。

----------

交换网口后直连和代理均能稳定在 100 ~ 120 M

kiddin9, 2022-06-13:
装个 luci-app-homebox 内网测速试试

gee1k, 2022-06-16:
上次测试的时候，用 iperf 正向、反向分别测试了，是没问题的。

我再用这个插件试试

gee1k, 2022-06-16:
luci-app-homebox 测试正常

[image]

SpeedTest
[image]

交换网口后

[image]

gee1k, 2022-06-16:
但是我刚刚升级了最新版固件后，在不交换网口的情况下直连测速的上传速度好像有所提升，之前是 10M 左右。现在能跑到 50 ~ 80 左右，有时候也能上 100M。代理情况下还是不行。

所以也不知道是不是 OpenClash 的问题，但是交换网口后都没问题，所以很奇怪

closed 2023-03-15 by gee1k