kiddin9/Kwrt #750: BUG Softethervpn5运行大概3分钟左右突然断开，提示网络设备丢失，等待20秒左右恢复。

krishuang0610, 2022-06-15:
Wed Jun 15 22:43:12 2022 user.notice firewall: Reloading firewall due to ifup of VPN (vpn_vpn)
Wed Jun 15 22:43:13 2022 daemon.info netdata[6099]: RRDSET: chart name 'net_packets.vpn_vpn' on host 'OpenWrt' already exists.
Wed Jun 15 22:43:43 2022 daemon.notice netifd: Network device 'vpn_vpn' link is down
Wed Jun 15 22:43:43 2022 daemon.notice netifd: Interface 'VPN' has link connectivity loss
Wed Jun 15 22:43:43 2022 kern.info kernel: [ 9819.164843] traps: vpnclient[2108] general protection fault ip:7f199b2a1b29 sp:7f199a784710 error:0 in libc.so[7f199b290000+4b000]
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): udhcpc: SIOCGIFINDEX: No such device
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): udhcpc: received SIGTERM
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): udhcpc: unicasting a release of 192.168.30.10 to 192.168.30.1
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): udhcpc: sending release
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): udhcpc: can't bind to interface vpn_vpn: No such device
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): udhcpc: bindtodevice: No such device
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): udhcpc: entering released state
Wed Jun 15 22:43:43 2022 daemon.notice netifd: VPN (2134): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "VPN" } (Permission denied)
Wed Jun 15 22:43:43 2022 daemon.notice netifd: Interface 'VPN' is now down
Wed Jun 15 22:43:43 2022 daemon.notice netifd: Interface 'VPN' is disabled
Wed Jun 15 22:43:48 2022 daemon.notice netifd: Interface 'VPN' is enabled
Wed Jun 15 22:43:48 2022 daemon.notice netifd: Network device 'vpn_vpn' link is up
Wed Jun 15 22:43:48 2022 daemon.notice netifd: Interface 'VPN' has link connectivity
Wed Jun 15 22:43:48 2022 daemon.notice netifd: Interface 'VPN' is setting up now
Wed Jun 15 22:43:48 2022 kern.info kernel: [ 9824.289256] device vpn_vpn entered promiscuous mode
Wed Jun 15 22:43:48 2022 daemon.notice netifd: VPN (2974): udhcpc: started, v1.35.0
Wed Jun 15 22:43:48 2022 daemon.notice netifd: VPN (2974): udhcpc: broadcasting discover
Wed Jun 15 22:43:49 2022 daemon.info netdata[6099]: RRDSET: chart name 'net.vpn_vpn' on host 'OpenWrt' already exists.
Wed Jun 15 22:43:49 2022 daemon.info netdata[6099]: RRDSET: chart name 'net_operstate.vpn_vpn' on host 'OpenWrt' already exists.
Wed Jun 15 22:43:49 2022 daemon.info netdata[6099]: RRDSET: chart name 'net_carrier.vpn_vpn' on host 'OpenWrt' already exists.
Wed Jun 15 22:43:49 2022 daemon.info netdata[6099]: RRDSET: chart name 'net_mtu.vpn_vpn' on host 'OpenWrt' already exists.
Wed Jun 15 22:43:49 2022 daemon.notice netifd: VPN (2974): udhcpc: broadcasting select for 192.168.30.10, server 192.168.30.1
Wed Jun 15 22:43:50 2022 daemon.notice netifd: VPN (2974): udhcpc: lease of 192.168.30.10 obtained from 192.168.30.1, lease time 7200
Wed Jun 15 22:43:50 2022 daemon.notice netifd: Interface 'VPN' is now up
Wed Jun 15 22:43:50 2022 user.notice firewall: Reloading firewall due to ifup of VPN (vpn_vpn)
Wed Jun 15 22:43:51 2022 daemon.info netdata[6099]: RRDSET: chart name 'net_packets.vpn_vpn' on host 'OpenWrt' already exists.
Wed Jun 15 22:44:11 2022 daemon.info uwsgi-luci: Wed Jun 15 22:44:11 2022 - worker 2 (pid: 1744) is taking too much time to die...NO MERCY !!!
Wed Jun 15 22:44:12 2022 daemon.info uwsgi-luci: uWSGI worker 2 cheaped.
Wed Jun 15 22:48:28 2022 daemon.notice netifd: Network device 'vpn_vpn' link is down
Wed Jun 15 22:48:28 2022 daemon.notice netifd: Interface 'VPN' has link connectivity loss
Wed Jun 15 22:48:28 2022 kern.info kernel: [10104.184714] traps: vpnclient[2963] general protection fault ip:7fa350577b29 sp:7fa34fa5a710 error:0 in libc.so[7fa350566000+4b000]
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): udhcpc: SIOCGIFINDEX: No such device
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): udhcpc: received SIGTERM
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): udhcpc: unicasting a release of 192.168.30.10 to 192.168.30.1
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): udhcpc: sending release
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): udhcpc: can't bind to interface vpn_vpn: No such device
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): udhcpc: bindtodevice: No such device
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): udhcpc: entering released state
Wed Jun 15 22:48:28 2022 daemon.notice netifd: VPN (2974): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "VPN" } (Permission denied)
Wed Jun 15 22:48:28 2022 daemon.notice netifd: Interface 'VPN' is now down
Wed Jun 15 22:48:28 2022 daemon.notice netifd: Interface 'VPN' is disabled

goven, 2022-07-22:
Softethervpn5能用吗？我设置好了怎么连不上呢？LEDE上的Softethervpn是可以用的。

krishuang0610, 2022-07-22:
能获取到IP也能连接上服务器，就是虚拟网卡不定时重启……导致基本无法使用。

获取 Outlook for iOS<https://aka.ms/o0ukef>

goven, 2022-07-22:
> 能获取到IP也能连接上服务器，就是虚拟网卡不定时重启……导致基本无法使用。
> 
> 
> 
> 获取 Outlook for iOS<https://aka.ms/o0ukef>
> 
> 
> 发件人: goven ***@***.***>
> 
> 发送时间: Friday, July 22, 2022 10:47:44 PM
> 
> 收件人: kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1 ***@***.***>
> 
> 抄送: krishuang0610 ***@***.***>; Author ***@***.***>
> 
> 
> 
> 
> 
> 
> Softethervpn5能用吗？我设置好了怎么连不上呢？LEDE上的Softethervpn是可以用的。

请问你是怎么设置的呢？我直接不能连上。但是在LEDE版本的固件上可以连接，在这个固件上就连不上。我QQ:1021073979，能指导一下吗，谢谢！

krishuang0610, 2022-07-22:
有没有什么错误提示？

goven, 2022-07-22:
提示服务器没有响应，我想用这个固件，但是用不了softethervpn。虽然在LEDE可以用，但是感觉LEDE没有这个固件清爽。麻烦指导一下设置方式，谢谢了！